mjansen4857/pathplanner #207: Red Alliance - Flip Path Along Y Axis

hadley31, 2023-01-08:
Add alliance parameter to PPLib classes, telling the path follower to flip the path along the Y axis

hadley31, 2023-01-08:
Third times the charm 🙂

hadley31, 2023-01-08:
Or possibly have the UI export two different paths, either in the same file or separate

mjansen4857, 2023-01-08:
I ink I want to avoid doing this in the UI if possible since it is a lot more effort to do it in a nice way. And exporting 2 different paths may be annoying to work with. I think doing it in the load path methods is really simple and the way to go.

WarrenReynolds, 2023-01-08:
I think this needs a lot more thought.  This isn't just a case of added a negative sign.  You are going to have to pass in the field width and alliance colour to the generate path method because coordinates of points and spine handles are all going to turn into (FieldWidth-Value) instead of Value. 

What have WPILib done with the whole single mirror instead of double mirror field setup thing. Does any body know how the coordinates of the April Tags are returned from the WPI Call?   Does this call take an Alliance colour or is Tag ID 1 always reported at the same location regardless of your alliance colour?

mjansen4857, 2023-01-08:
I think it’s safe to assume that everyone using 2023 PPLib will be playing on the charged up field so there’s not really a need to take in field width and what not. Idk what WPILib will do with pathweaver but I haven’t seen any updates to it since last year. 

The WPILib apriltags stuff has the ability to take alliance color into account when returning tag positions. However, this was likely designed with a normal field mirror so it will probably be updated.

gcschmit, 2023-01-08:
AprilTags are unique on the field (no ID is repeated). As a result, AprilTags themselves aren't directly impacted by the reflected field layout.

hadley31, 2023-01-08:
I thought about that too, but it took me a while to realize that the AprilTagFieldLayout will work no matter what. The positions will always be correct since they are just _relative to_ different points. It is just the autos that are affected.

WarrenReynolds, 2023-01-09:
This all hinges around do you call 0,0 your alliance wall on the right side regardless of if you are Red or Blue?

In Regard to the April tags, here is my take. The April Tag Pose3D's are unique for each April Tag ID on the field. So in WPILib eyes, 3 of them have a physical X-Coord of approx 1m (This could be either Tag IDs 6,7,8 or 1,2,3) with an orientation of 0 zero degrees (ie Facing down the field) and the other 3 tags (either 1,2,3 or 6,7,8) have an X- Coord of approx 15.5m with an orientation of 180 degrees. (Facing back towards the middle of the field). I'm talking about the tags on shelves that hold the cubes here, not the 2 tags on the loading stations.

 I can't really tell from the SetOrigin source code, which is which, as the Enum OriginPosition is a bit ambiguous to me.  Is the right in kBlueAllianceWallRightSide when viewed from the scoring table side (ie this means you are Red Alliance) or is it when viewed from the other side of the field (ie this mean you are Blue Alliance)

Anyway, once you work out what parameter to call SetOrigin with, a call to GetTagPose(ID) will give you different values.

```
void AprilTagFieldLayout::SetOrigin(OriginPosition origin) {
  switch (origin) {
    case OriginPosition::kBlueAllianceWallRightSide:
      SetOrigin(Pose3d{});
      break;
    case OriginPosition::kRedAllianceWallRightSide:
      SetOrigin(Pose3d{Translation3d{m_fieldLength, m_fieldWidth, 0_m},
                       Rotation3d{0_deg, 0_deg, 180_deg}});
      break;
    default:
      throw std::invalid_argument("Invalid origin");
  }
}
```
Every field picture in the manual has the scoring table at the bottom of the image and the Blue Alliance on the left as the default.  Going on the source code above it looks like the actual April Tag Pose3D are set with a Blue Alliance Wall on the right and when the Red Alliance Wall is on the right all the Tag Poses are translated and rotated.  Seems oopiste to what it should be to me.

I suppose where I'm going with all this is, if you are going to use the advanced features of PathPlanner like firing commands at stop points or points along the path and if these commands are going to use April tag ID's to do last minute alignments before you score an item you will either need some way to uses Different April Tag IDs in these commands that are called depending if you are red or blue alliance or alternatively you will need to build completely different paths that have different commands to execute that have hard coded April Tag ID's in them depending if you are Red or Blue.

Maybe all the commands that are used in Autos will need the ability to query alliance colour to work out how they will behave, with every command having a case statement to handle the different alliance colours.

 For example in your two cube auto you correct alignment to TagID 7 if you are Blue and correct your alignment to TagID 2 if you are Red before your score your second element.

Food for thought.

hadley31, 2023-01-09:
You should always assume the right corner of your alliance station is 0,0. I agree the enum value names are a little confusing. It does not mean that the blue alliance station is on the right side of the image, or that the red alliance station is on the right side of the image. What they are referring to is, the origin point is either _the right corner of the blue alliance station_, or _the right corner of the red alliance station_. As long as you set the correct origin, there is nothing else you need to do. The poses will be correct based where they are relative to the right corner of your alliance station.

WarrenReynolds, 2023-01-09:
FYI 
Just found the Tags JSON File [https://github.com/wpilibsuite/allwpilib/blob/main/apriltag/src/main/native/resources/edu/wpi/first/apriltag/2023-chargedup.json](https://github.com/wpilibsuite/allwpilib/blob/main/apriltag/src/main/native/resources/edu/wpi/first/apriltag/2023-chargedup.json)

WarrenReynolds, 2023-01-09:
> What they are referring to is, the origin point is either _the right corner of the blue alliance station_, or _the right corner of the red alliance station_.

Thanks, that makes so much more sense now. After looking at the actual values for the locations of the Tags in the Json file, I think the Blue and Red are need to be swapped.  But that's nothing to do with PathPlanner, thats a WPILib issue.

mjansen4857, 2023-01-09:
So the solution I've come up with is pretty simple and should require no change from the user perspective as long as they are using the PPLib path following commands. Instead of generating a second path, the path states are just transformed depending on the alliance when the path is sampled (with a method overload in case some people don't want that).

I believe I changed everything I need to (robot Y position and anything curvature/rotation related), but here is the code I'm using for this just to have some more eyes on it to see if I missed anything before I try releasing it:
```Java
private PathPlannerState transformForAlliance(
      PathPlannerState state, DriverStation.Alliance alliance) {
    if (alliance == DriverStation.Alliance.Red) {
      Translation2d transformedTranslation =
          new Translation2d(state.poseMeters.getX(), FIELD_WIDTH_METERS - state.poseMeters.getY());
      Rotation2d transformedHeading = state.poseMeters.getRotation().times(-1);
      Rotation2d transformedHolonomicRotation = state.holonomicRotation.times(-1);

      state.poseMeters = new Pose2d(transformedTranslation, transformedHeading);
      state.angularVelocityRadPerSec *= -1;
      state.holonomicRotation = transformedHolonomicRotation;
      state.holonomicAngularVelocityRadPerSec *= -1;
      state.curveRadius *= -1;
      state.curvatureRadPerMeter *= -1;
    }

    return state;
  }
```

We-Gold, 2023-01-09:
With the apriltag absolute positioning this year, will this create a conflict because the initial pose will be relative to the blue side?

mjansen4857, 2023-01-09:
You can set the origin of the WPILib apriltag stuff to the red side of the field if you are on the red alliance, and the apriltag positions will be transformed based on that new origin. This issue is about essentially doing the same thing with the paths, so it is to avoid coordinates conflict.

closed 2023-01-10 by mjansen4857

edf42001, 2023-01-18:
One downside to this method is that the Field2D display always has the origin at the bottom left corner on the blue alliance side. A path that has been transformed from the blue to red alliance will still be displayed on the field on the blue alliance side, so the path will not line up with the field image.

This would also mess with a vision simulation system. A path that starts facing apriltag id 8 on the blue alliance, in competition, when on the red alliance, will start facing tag 1. `AprilTagFieldLayout.getTagPose()` will compensate for this using `OriginPosition.kRedAllianceWallRightSide`.  But in simulation, the origin remains on the blue alliance side, so the robot will start facing tag 5, which mistakenly puts the robot up near the double substation.

A possible alternative, instead of reflecting across the center x-axis, then rotating the origin point by 180 degrees about the center, would be to reflect paths around the center y-axis, keeping the only origin as the blue alliance. (Group theory says this is the same!) 
[image]

Because the origin would stay on the blue alliance side no matter what alliance the robot is on, apriltag poses wouldn't need to be converted, but the robot's starting pose would also need to be transformed based on alliance.

I think this could be something to consider next time the field isn't rotationally symmetric.

mjansen4857, 2023-01-18:
> But in simulation, the origin remains on the blue alliance side

Why is that? You still have access to an alliance color dropdown in simulation. The only thing with the origin still on blue alliance side is the Field2d widget, and that is really only due to the field image. If you don't use a field image, or use a flipped field image then you don't really need to worry about that.

> A possible alternative... would be to reflect paths around the center y-axis, keeping the only origin as the blue alliance

This is a possibility, but it goes against the standard FRC assumption that (0, 0) is on your side of the field. Flipping the paths like this would mean that a lot of teams would have to make code changes to support it. For example, field oriented drive would need to be flipped depending on alliance color as well, since a positive X axis movement would now be towards the red alliance wall instead of away from it like it would be if (0, 0) was on the red side.

But, by flipping the path across the x axis and assuming (0, 0) is always on your side of the field, teams don't need to make any changes to their code to support a different coordinate system and can just let PPLib transform the path as needed.

edf42001, 2023-01-18:
> If you don't use a field image, or use a flipped field image then you don't really need to worry about that.
Good point, it wouldn't be too hard to make a flipped field image

> You could use AdvantageScope to connect to the simulator which lets you choose which side of the field the origin is on.
Sounds cool, will have to look into it

I didn't know that (0, 0) on your side of the field was the standard FRC assumption, and forgot about how that would affect field-centric drive. You've convinced me, it would be more trouble than it's worth to change. To solve my very specific issue of "apriltags in simulation use the wrong reference frame", perhaps I could poll `DriverStation.getAlliance()` and update apriltag positions accordingly.

Thanks for the input!

gcschmit, 2023-01-18:
> I didn't know that (0, 0) on your side of the field was the standard FRC assumption, and forgot about how that would affect field-centric drive. You've convinced me, it would be more trouble than [...]

We do exactly that (layout references the AprilTagFieldLayout object):

```
if (DriverStation.getAlliance() != lastAlliance) {
      lastAlliance = DriverStation.getAlliance();
      if (DriverStation.getAlliance() == DriverStation.Alliance.Red) {
        layout.setOrigin(OriginPosition.kRedAllianceWallRightSide);
      } else {
        layout.setOrigin(OriginPosition.kBlueAllianceWallRightSide);
      }
```

We still are forgetting to change the driver station color in the simulator, but we will learn!