mjansen4857/pathplanner #210: PathPlanner Immediately quits itself on M1 Mac

team3572, 2023-01-10:
Downloaded from App Store.
Launch, see a window for a micro second then the app quits.
I don't think it crashes because I don't get a crash window.

I had an old version installed from 2020, it is possible there is a cache folder or something I'd need to delete to get the new version running?

team3572, 2023-01-10:
Some messages from the console...
error	11:28:16.470345-0500	PathPlanner	cannot open file at line 45530 of [9ff244ce07]
error	11:28:16.470355-0500	PathPlanner	os_unix.c:45530: (2) open(/var/db/DetachedSignatures) - No such file or directory
default	11:28:16.479386-0500	PathPlanner	Entering exit handler.

mjansen4857, 2023-01-10:
I'm able to launch it when installed from the app store so not sure what's wrong on your end. Does it work if you download from github?

team3572, 2023-01-11:
So after getting a bunch of stuff installed (ruby, cocoapods, Flutter). I was able to build it from the repository using Flutter and the build version worked.
For me, the app store version still simply quits.
So I'm ok with this since I can getting it working but if you want anymore info from me on the app store version like a log or something please let me know.

mjansen4857, 2023-01-11:
Yeah I can't really find anything relevant from the console output you provided. There was an update yesterday, did you update your app store version to that?

Also, there are mac builds in the github releases and github actions if you prefer to not build it manually all the time.

cpostbitbuckets, 2023-01-14:
One of my students was having this problem. Pathplanner was crashing on his M1 while trying to read secure bookmarks.

https://github.com/mjansen4857/pathplanner/blob/main/lib/pages/home_page.dart#L79

We deleted the preferences from last season's pathplanner by executing this in the terminal: `defaults delete com.mjansen4857.pathplanner`

Now it launches.

mjansen4857, 2023-01-14:
@team3572 perhaps give that a try. It may be a different issue if your most recent version before this one was from 2020, which didn’t use secure bookmarks or defaults at all. Worth a try though.

aliddy23, 2023-01-15:
> We deleted the preferences from last season's pathplanner by executing this in the terminal: `defaults delete com.mjansen4857.pathplanner`

This worked for me as well.

team3572, 2023-01-15:
Worked. Thanks.

mjansen4857, 2023-01-15:
Awesome. I think I know what is specifically causing the crash, so I will leave this issue open to try and catch that issue to prevent it from crashing. It happens when the project you previously had open doesn't exist anymore. Fixed this issue earlier for other platforms but macOS needs to access secure bookmarks to read the files in the directory so it can crash there too I guess.

Chris2fourlaw, 2023-01-20:
Unable to reproduce with MacBook Pro 16 in, 2021, M1 Pro

Savimaster, 2023-04-07:
> 

This worked for me as well

Savimaster, 2023-04-07:
> One of my students was having this problem. Pathplanner was crashing on his M1 while trying to read secure bookmarks. [...]

This also worked for me.

PR #317 (2023-08-08): 2024 Reworks
Majority of the reworks for the 2024 release.

Fixes #316, closes #311, closes #304, fixes #301, closes #289, closes #285, closes #274, closes #260, fixes #226, fixes #210, closes #163, closes #138, closes #126

Supersedes #300 

TODO:
- [X] GUI reworks
- [X] Java PPLib reworks
- [X] C++ PPLib reworks (missing pathfinding for now)
- [X] Test Java PPLib in robot project
- [x] Documentation
- [x] Finish GUI unit tests
- [x] Test (and probably fix) C++ PPLib in robot project
closed 2023-08-08 by mjansen4857