mjansen4857/pathplanner #214: Issue with Y value in resetPose

EddieDL, 2023-01-14:
I know there is an update that isn't yet in a release to fix issues where the red alliance pose reset is not correct. However, I am having a problem on the Blue side.

When I use the attached path, the resetPose call is passed a pose with the following values:
x: 1.8
y: 7.59
rot: 0

I believe I have configured the auto path on the blue alliance, but am confused why it is passing a different y than what is configured in the path file.

Perhaps the latest update fixes this issue as well?

[Uploading Straight Test.json…]()

EddieDL, 2023-01-14:
{
  "waypoints": [
    {
      "anchorPoint": {
        "x": 1.8,
        "y": 0.43
      },
      "prevControl": null,
      "nextControl": {
        "x": 2.8,
        "y": 0.43
      },
      "holonomicAngle": 0.0,
      "isReversal": false,
      "velOverride": null,
      "isLocked": false,
      "isStopPoint": false,
      "stopEvent": {
        "names": [],
        "executionBehavior": "parallel",
        "waitBehavior": "none",
        "waitTime": 0
      }
    },
    {
      "anchorPoint": {
        "x": 2.8,
        "y": 0.43
      },
      "prevControl": {
        "x": 1.7999999999999998,
        "y": 0.43000000000000016
      },
      "nextControl": null,
      "holonomicAngle": 180.0,
      "isReversal": false,
      "velOverride": null,
      "isLocked": false,
      "isStopPoint": false,
      "stopEvent": {
        "names": [],
        "executionBehavior": "parallel",
        "waitBehavior": "none",
        "waitTime": 0
      }
    }
  ],
  "markers": []
}

mjansen4857, 2023-01-14:
The update was released. 2023.3.1.

Did you set your alliance color to blue in the driver station?

EddieDL, 2023-01-14:
From what I can tell the latest version is missing 2 commits. One of which is the fix for the pose reset.

No matter what side I have selected the y value passed in is the same.

mjansen4857, 2023-01-14:
The releases on this repo are unrelated to PPLib releases. 

Anyways, found the issue in the code. Will push another update tonight

EddieDL, 2023-01-14:
Oh I didn't realize that. Good to know.

Thanks for looking at it quickly. I am just happy that it means my code might not be broken :)

closed 2023-01-14 by mjansen4857

mjansen4857, 2023-01-14:
Ok update pushed. Are you able to confirm if the problem is fixed?

EddieDL, 2023-01-15:
I can't check tonight. I might be able to test tomorrow. If not it will be Monday :( Sorry

mjansen4857, 2023-01-15:
No problem. Just want to make sure that the issue that was fixed is what was causing your problem. Pretty sure it is, but I don’t want to leave you hanging if not.

EddieDL, 2023-01-17:
I can confirm that your latest fix addressed my issue. Thanks for investigating and resolving.