mjansen4857/pathplanner #218: Add a feature in the GUI to flip paths from the code

Mason-Beard, 2023-01-15:
Add a feature that allows current paths to be copied and/or flipped from blue to red and vise-versa

mjansen4857, 2023-01-15:
PathPlannerLib already handles this, since doing it in the GUI is just not the way to go for a couple reasons:
* You end up needing to change 2 paths instead of 1
* (0, 0) should always be on your side of the field. PathPlanner (0, 0) is always on the blue side, so if you made a path on the red side, the coordinates would be wrong.

You should just ignore alliance colors in the GUI entirely and let PPLib transform the path depending on the alliance color, the red side is really only there for visualization.

Feel free to discuss more / ask questions if you have any, but I will close this since it won't be worked on.

closed 2023-01-15 by mjansen4857

Mason-Beard, 2023-01-16:
We want to have easy access to creating a flipped version of our auto in the event that the field's construction is off for one side at a particular event.

mjansen4857, 2023-01-16:
Since I’ve started doing FRC I’ve never actually run in to a situation where that was necessary, but maybe we’ve just been lucky.

But still, you wouldn’t want to flip this to the red side of the field since the coordinates would be wrong. If you need to do this, you can just duplicate the path (this functionality is already in the GUI) and then make your adjustments to the duplicated path, then transform it for the red alliances with PPLib.