mjansen4857/pathplanner #223: deploy updated path to roboRIO without re-deploying robot code

gcschmit, 2023-01-21:
When iterating on an auto path, it would be useful to be able to deploy the updated path to the roboRIO without re-deploying the robot code and waiting for that process to complete. In our robot code, we would need to reload our paths as that is usually only done when the RobotContainer is constructed, but that additional complexity would probably be worth the trouble. (I naively thought that PathPlannerServer helped with this, but now that I took the time to read the wiki I know better.)

Probably not something for this season but perhaps to consider in the future.

mjansen4857, 2023-01-21:
So you're asking for the paths to regenerated when PathPlannerServer updates them so you don't need to reload code? Because it already handles deploying updated paths so I'm assuming that's what you mean.

gcschmit, 2023-01-21:
Sounds like we need to spent more time with PathPlannerLib Server. If we enable PathPlannerLib Server and connect PathPlanner to it, and change a path in PathPlanner, is that change reflected on the robot and would be reflected the next time we run an auto path? (We subclass PPSwerveControllerCommand.)

gcschmit, 2023-01-21:
We enabled PathPlannerLib Server and explored it this morning. It is awesome and does exactly what we want. I was focused on the visualization of the trajectory and pose and missed the deploy feature.

closed 2023-01-21 by gcschmit