mjansen4857/pathplanner #224: Add wheel speed setpoints / ChassisSpeeds to network tables

edf42001, 2023-01-21:
In `PPRamseteCommand`, do you think it would be useful to add `leftSpeedSetpoint`, `rightSpeedSetpoint`, as well as the `ChassisSpeeds`' `vxMetersPerSecond`, `vyMetersPerSecond`, and `omegaRadiansPerSecond` to SmartDashboard? This would allow teams to plot these setpoints vs the actual values measured from the robot, in order to verify and tune feedforward / DiffDriveKinematics gains and velocity PID loops.

mjansen4857, 2023-01-21:
I think it’s best to leave this up to the user, since these values are already passed through the output methods. So, you could just add these set points to SD in your output methods.

edf42001, 2023-01-22:
Which output methods? I tried to find a way to access the values, but they look to be locked up inside the implementation of the command (such as on these lines): https://github.com/mjansen4857/pathplanner/blob/00920e70c41642f82fae201eec277cf79190daf0/pathplannerlib/src/main/java/com/pathplanner/lib/commands/PPRamseteCommand.java#L292-L293

mjansen4857, 2023-01-22:
You should be able to use your kinematics to convert from left and right speeds to ChassisSpeeds (if diff. drive kinematics are anything like SwerveDriveKinematics, not 100% sure on that)

I am in the middle of working on a way to add custom logging callbacks to the path following commands, so I can potentially add another one for ChassisSpeeds setpoints too

edf42001, 2023-01-22:
Thanks for the workaround. Having the ability to access these setpoints without duplicating code or backtracking calculations will be great. I look forward to the custom callbacks.

closed 2023-01-22 by edf42001