mjansen4857/pathplanner #225: BaseAutoBuilder::resetPose is crashing in c++ fullauto

WarrenReynolds, 2023-01-22:
This could just be a problem from somewhere else in our code but if we try to run a Full Auto in crashes Rio on execution of the autocommand..
I have narrowed it down to following code inside resetPose when m_useAllianceColor is true.

```
						initialState =
								PathPlannerTrajectory::transformStateForAlliance(
										initialState,
										frc::DriverStation::GetAlliance());
```
If I comment out this code, the Auto works. (IE The pose is passed to our resetPose function)  If I uncomment this code, it crashes the robot code when executing the auto.
Is it something to do with the variable scope of the lambda function.  Should the initialState calculation be moved above the above the creating of the RunOnce Command.

mjansen4857, 2023-01-22:
It needs to be in the command or else it would transform based on the alliance color when the command is created. I’ll play around with it and have a look. 

Does it work fine if useAllianceColor is false? I guess just leave it false for now if so

mjansen4857, 2023-01-23:
What versions of PPLib and WPILib are you using? I can't reproduce on WPILib 2023.2.1 and PPLib 2023.3.4

WarrenReynolds, 2023-01-24:
Not exactly sure how to check the WPILib version but 99% sure I installed 2023.2.1 only a few days ago.
The PPLib was downloaded from GitHub and copied into the project locally on 22/1/2023. Not sure what version that makes it.

For reference , the call to frc::DriverStation;;GetAllaince() is what is causing our code to crash. The full auto Command Group is put together when auto starts, and then it is scheduled directly after it is made. Both steps are happening at run time. So for us it doesn't have to be in the lambda.

 If the GetAlliance() call is part of the RunOnce lambda used to create the resetPose command, it crashes, If I move it above the lambda, and stores its value in a local variable and then add that variable to capture [] for the lamba, it works.  I don't understand why this is this case.  

NB: I realise why its in the lambda for people that are constructing multiple FullAuto command group at robot code startup, but for us we are only using the one FullAuto command and we are constructing it at runtime.

PeterJohnson, 2023-01-25:
Make sure you're not destroying the AutoBuilder before the command runs.  AutoBuilder is doing a lambda capture of `this`, so if the AutoBuilder is destroyed before the command runs, you're in UB territory with a use-after-free.  See the discussion here: https://www.chiefdelphi.com/t/path-planner-ramseteautobuilder-in-c-robot-crashing/424598/3?u=peter_johnson

WarrenReynolds, 2023-02-12:
I've been talking about this issue out on the CD thread that @PeterJohnson mentioned above.
[https://www.chiefdelphi.com/t/path-planner-ramseteautobuilder-in-c-robot-crashing/424598/3?u=peter_johnson](https://www.chiefdelphi.com/t/path-planner-ramseteautobuilder-in-c-robot-crashing/424598/3?u=peter_johnson)

When we moved the Autobuilder object out to the robotcontainer as a member variable so that it remained in scope, this stopped the code crashing the Rio, but for some unknown reason, now the command group returned by .fullAuto(...) wasn't requiring the drive subsystem anymore and this was breaking the path following because the default drive command was executing at the same time as the path following. (causing serious jitters)  I'm not saying this is a problem with pathplanner, It's probably just the way we are trying to implement it into our code.  But I thought I'd include this information here for others that might be having similar issues with c++.

mjansen4857, 2023-02-12:
Is your drive subsystem being fully created before the auto builder is created? Not sure why it isn't setting the requirements correctly if you're still passing the subsystem to it.

WarrenReynolds, 2023-02-14:
I'm not sure as they are both being created in the robotcontainer and I'm not sure how all this plays out in terms of execution. IE: In the class header definition verses in the class constructor after the : or in the class constructor code block between the {}

Recap.
The reason we had to create the SwerveAutoBuilder object as a member variable to the robot container was because the SwerveAutoBuilder class is based on the BaseAutoBuilder class, and inside the BaseAutoBuilder class there is a resetPose method that captures [this] in its code to build a command to reset the pose. So the BaseAutoBuilder object (which implies the SwerveAutoBuilder object) must stay in scope until the command it creates is done. Otherwise it crashes the Rio

In order to create a SwereAutoBuilder you need to supply it few things that are related to the drive subsubsystem.  Namely the GetPose, ResetPose, SetChassisSpeeds and a pointer to the drive subsystem so the requirements work.

Currently in the header for the RobotContainer Object we have.

  `DriveSubsystem m_drive;`
Followed by 
` pathplanner::SwerveAutoBuilder *m_autoBuilder;`

Then in the constructor for the robot container we have
```
RobotContainer::RobotContainer() {
    
    fmt::print("Adding commands to the Command Map\n");
    m_eventMap.emplace("marker1", std::make_shared<frc2::PrintCommand>("Passed Marker 1"));
    m_eventMap.emplace("marker2", std::make_shared<frc2::PrintCommand>("Passed Marker 2"));
    m_eventMap.emplace("marker3", std::make_shared<frc2::PrintCommand>("Passed Marker 3"));
    m_eventMap.emplace("marker4", std::make_shared<frc2::PrintCommand>("Passed Marker 4"));
    

    
    m_autoBuilder = new pathplanner::SwerveAutoBuilder(
        [this]() { return m_drive.GetPose(); }, // Function to supply current robot pose
        [this](auto initPose) { m_drive.PrintPose(initPose); }, // Function used to reset odometry at the beginning of auto

         /* nb: we removed this resetPose call from autoBuilder and did the reset another way.
          * We start with our own InitializeOdometry command, 
          * as we have learned (the hard way) to set and CHECK all odometry resets.
          * ie occasionally, telling an encoder or a pigeon to reset fails (which leads to bad news), 
          * so now we repeat setting it until we get back the same values that we set.
          * before we move onto to following paths
          */
        PIDConstants(5.0, 0.0, 0.0), // PID constants to correct for translation error (used to create the X and Y PID controllers)
        PIDConstants(2.0, 0.0, 0.0), // PID constants to correct for rotation error (used to create the rotation controller)
        [this](auto speeds) { m_drive.driveFieldRelative(speeds); }, // Output function that accepts field relative ChassisSpeeds
        m_eventMap, // Our event map
        { &m_drive }, // Drive requirements, usually just a single drive subsystem
        true // Should the path be automatically mirrored depending on alliance color. Optional, defaults to true
   );
```

As far as I can tell the Drive subsystem is fully created because the three (GetPose, ResetPose, SetChassisSpeeds) all work when we execute the fullauto command. We created the SwerveAutoBuilder object as late as possible by heap allocating it (using "new").

Maybe the "drive subsystem" isn't being fully created until after the robot Container Object is fully created (ie after the robotcontainer constructor code terminates).  Again I'm not sure of the order of creation of objects in class definitions that contain class definitions that contain class definition.

I would have thought it went something like this
```
Start creating Robot Object
    Start creating RobotContainer Object
        Start creating DriveSubsystem Object
            Start Creating SwerveModuleFL Object
            Finish Creating SwerveModuleFL Object
            Start Creating SwerveModuleFR Object
            Finish Creating SwerveModuleFR Object
            Start Creating SwerveModuleRL Object
            Finish Creating SwerveModuleRL Object
            Start Creating SwerveModuleRR Object
            Finish Creating SwerveModuleRR Object
        Finish creating DriveSubsystem Object
    Finish creating RobotContainer Object
Finish creating Robot Object
```
We have a workaround to require the drive subsystem by other means so we can move on, but I'd really like to know why its not working as its supposed to.  

Could the following be the cause of the problem
NB: Please note the different case of the F letter on the start of FollowPathWithEvent
FollowPathWithEvents.cpp
has the following line in it
`	this->AddRequirements(m_pathFollowingCommand->GetRequirements());`

and the **f**ollowPathWithEvents method of the BaseAutoBuilder returns a **F**ollowPathWithEvents object.  I assume the "return" statement is basically converting the command created by **F**ollowPathWithEvents to the new style commandPtr and returning that. But doesn't the **F**ollowPathWithEvents object created in the return statement effectivity go out of scope after the return statement has finished, and hence the [this] pointers used inside the **F**ollowPathWithEvents object will be invalid.

I'm probably barking up the wrong tree as I'm still trying to get my head around [this] pointers.

WarrenReynolds, 2023-02-14:
Scape that last bit about [this].  There is no lambda captures on [this] so I'm completely wrong on this chain of thought.

mjansen4857, 2023-08-09:
Going to close this, but the new reworks should make this stuff a lot easier to avoid.

AutoBuilder is now a static class that you just configure once and it gets used internally. So basically, call `AutoBuilder::configureHolonomic` or some other variation in the constructor of your drive subsystem and then you don't have to worry about anything going out of scope or holding on to the auto builder object. *Hopefully.*

closed 2023-08-09 by mjansen4857