mjansen4857/pathplanner #226: Window overflows on Linux HIDPI configuration

jr1221, 2023-01-22:
Hello,

On linux, the window overflows on startup.

Error:
```
flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: │ A RenderFlex overflowed by 159 pixels on the bottom.
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ #0   Log.log (package:pathplanner/services/log.dart:40:15)
flutter: │ #1   Log.error (package:pathplanner/services/log.dart:63:5)
flutter: │ #2   main.<anonymous closure>.<anonymous closure> (package:pathplanner/main.dart:24:13)
flutter: │ #3   FlutterError.reportError (package:flutter/src/foundation/assertions.dart:1184:14)
flutter: │ #4   DebugOverflowIndicatorMixin._reportOverflow (package:flutter/src/rendering/debug_overflow_indicator.dart:239:18)
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ ⛔ Flutter Error
flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

======== Exception caught by rendering library =====================================================
The following assertion was thrown during layout:
A RenderFlex overflowed by 159 pixels on the bottom.

The relevant error-causing widget was: 
  Column Column:file:///home/jackr/Projects/StudioProjects/pathplanner/lib/pages/welcome_page.dart:72:23
The overflowing RenderFlex has an orientation of Axis.vertical.
The edge of the RenderFlex that is overflowing has been marked in the rendering with a yellow and black striped pattern. This is usually caused by the contents being too big for the RenderFlex.

Consider applying a flex factor (e.g. using an Expanded widget) to force the children of the RenderFlex to fit within the available space instead of being sized to their natural size.
This is considered an error condition because it indicates that there is content that cannot be seen. If the content is legitimately bigger than the available space, consider clipping it with a ClipRect widget before putting it in the flex, or using a scrollable container rather than a Flex, like a ListView.

The specific RenderFlex in question is: RenderFlex#ddcde relayoutBoundary=up7 OVERFLOWING
...  parentData: offset=Offset(187.5, 0.0); flex=null; fit=null (can use size)
...  constraints: BoxConstraints(0.0<=w<=Infinity, 0.0<=h<=304.0)
...  size: Size(265.0, 304.0)
...  direction: vertical
...  mainAxisAlignment: center
...  mainAxisSize: max
...  crossAxisAlignment: center
...  verticalDirection: down
```
Screenshot of overflow: 
[image]

This many be due to my Hi-DPI configuration (3840x2160).  If I set the `maximumSize` to my resolution, I can resize the window larger and then nothing overflows.  I believe there should be a way to remedy this, as hardcoded resolution sizes are problematic for 4K displays (`main.dart:27`):
```dart
WindowOptions windowOptions = WindowOptions(
        size: const Size(1280, 720),
        minimumSize: const Size(640, 360),
        maximumSize: const Size(3840,2160),  // this change allows me to resize to my display
        center: true,
        title: 'PathPlanner',
        titleBarStyle:
            Platform.isMacOS ? TitleBarStyle.normal : TitleBarStyle.hidden,
      );
```
This is the solution I am using PathPlanner with.  Theoretically this should adapt to all screen sizes, but a report from the API that isn't 16:9 (which is possible) could allow the screen to resize larger than it is allowed to.  I really don't know the errors that could happen:
```dart
        maximumSize: WidgetsBinding.instance.window.physicalSize,
```

This is a great app, I wish I could open this issue with a solution that is a lot less hacky than this.

mjansen4857, 2023-01-23:
You're not able to resize the app to your screen with the main branch? There is no maximum size in the main branch so it should be able to resize to anything. I just tested it on my display and both manual sizing and the maximize button both work fine.

jr1221, 2023-01-23:
That does not seem to be working on Linux.  I will try Windows tommorow, perhaps it is a problem with window manager's linux code.  It is behaving such that the max size is the size (1280,720) when the field is not explicitly set, so maximizing doesn't do anything.  I cannot manually resize at all.  This is all in the main branch.

mjansen4857, 2023-01-23:
If you try updating window manager in pubspec.yaml 0.3.0 does that help at all?

jr1221, 2023-01-23:
Yes, it allows for the maximize button to work correctly, so the issue no longer keeps the whole app from being seen.  However there is still no ability to resize.  `isResizable()` returns true, I am guessing it is a window manager library problem.  I might open an issue there.

mjansen4857, 2023-01-23:
Yeah that sounds like a good idea. Feel free to open a PR to update window manager and set some big maximum size for now (probably 8K). I can always do it too if you'd prefer.

`WidgetsBinding.instance.window.physicalSize`
This line won't actually work as a work around, since it returns the size of the app window on the display, not the size of the display.

PR #317 (2023-08-08): 2024 Reworks
Majority of the reworks for the 2024 release.

Fixes #316, closes #311, closes #304, fixes #301, closes #289, closes #285, closes #274, closes #260, fixes #226, fixes #210, closes #163, closes #138, closes #126

Supersedes #300 

TODO:
- [X] GUI reworks
- [X] Java PPLib reworks
- [X] C++ PPLib reworks (missing pathfinding for now)
- [X] Test Java PPLib in robot project
- [x] Documentation
- [x] Finish GUI unit tests
- [x] Test (and probably fix) C++ PPLib in robot project
closed 2023-08-08 by mjansen4857