mjansen4857/pathplanner #237: C++ resetPose always sets rotation to zero

gcjurgiel, 2023-02-01:
When using `fullAuto` on a differential drive, the initial call to `resetPose` passes the correct translation of the initial state of the path to my provided lambda, but the rotation is always zero no matter the initial rotation of the path. This seems to occur both with and without useAllianceColor. I suspect I have configured something wrong since I have found nothing in the source that would cause this, but so far nothing I have done has fixed it either.

For reference, I am finding this discrepancy by comparing prints of the values returned from `pathGroup[0].getInitialState().pose` and what is passed into my `resetPose` lambda, which is strange since in this configuration those should be the same. Below are excerpts of the relevant code:

RobotContainer.h (private member variable declaration)

```
pathplanner::RamseteAutoBuilder autoBuilder {
    [this]() { return driveSubsystem.getPose(); }, [this](frc::Pose2d initPose) {
      std::cout << "resetPose: (" << units::length::to_string(initPose.X())
                << ", " << units::length::to_string(initPose.Y())
                << ", " << units::angle::to_string(initPose.Rotation().Degrees()) << ")" << std::endl;
      driveSubsystem.resetOdometry(initPose); 
    }, DriveConstants::ramseteController, DriveConstants::kinematics,
    [this](units::meters_per_second_t leftVelocity, units::meters_per_second_t rightVelocity) { 
      driveSubsystem.driveWheelSpeeds(leftVelocity, rightVelocity); 
    }, autoEventMap, {&driveSubsystem}, false
  };
```

RobotContainer.cpp (constructor)
```
auto pathGroup = pathplanner::PathPlanner::loadPathGroup("Auto", 1.5_mps, 2.0_mps_sq, false);
auto pose = pathGroup[0].getInitialState().pose;

std::cout << "Initial Path Pose: (" << units::length::to_string(pose.X())
          << ", " << units::length::to_string(pose.Y())
          << ", " << units::angle::to_string(pose.Rotation().Degrees()) << ")" << std::endl;

autoCommand = autoBuilder.fullAuto(pathGroup);
```

mjansen4857, 2023-02-01:
What version are you using?

gcjurgiel, 2023-02-01:
Version 2023.3.1 (20) of the GUI and version 2023.4.0 of the vendor library. WPILib is version 2023.2.1
Additionally I'm using the arm Mac version of the app.

mjansen4857, 2023-02-02:
Should keep issue open and link to the PR

closed 2023-02-02 by mjansen4857