mjansen4857/pathplanner #239: Swerve Rotation not occuring

Crossle86, 2023-02-06:
I am just starting with PP and have generated several paths testing them in simulation. Our swerve code works in sim and on our robot (but right now only sim available). Using PPSewerveControllerCommand, the sim robot follows the path as expected but holonomic rotation does not occur. The path preview shows  the rotation so it is in the path. PP seems simple and the setup seems simple but something is not right and I don't know what to do next to determine why no rotation. Any thoughts?

Also, the path as displayed in the sim is off a fair bit from what the PP editor shows, just looking at the starting point which appears left and down. The sim robot follows the path in general but stops well before the path stops in PP editor.

mjansen4857, 2023-02-07:
Can't really help much without seeing your code. There's a lot of places where a mistake can mess things up like this.

Crossle86, 2023-02-07:
Thanks for the quick reply. If  you are willing to look at the code you can find it here: https://github.com/ORF-4450/Robot23.

Our swerve code is based on SDS example code with modifications by us. The core swerve code is in our library: RobotLib. The Robot23 project has our swerve drive base code which calls into the RobotLib swerve functions. In the Robot23 project, there is a command in the autonomous folder called AutoDrivePPTrajectory. This is where we are following the PP path. This command is used by the auto command TestAuto4. TestAuto4 runs a path created right in that command or one loaded from a path file (happens bottom of RobotContainer).

Note: RobotLib is pulled from Jitpack but you have to change the version in the vendordep from "local" to "4.3.0".

Again, the PP path is followed but we don't get the holonomic rotation.

mjansen4857, 2023-02-08:
What if you use the ChassisSpeeds output override of PPSwerveController command instead of the one that outputs module states? It shouldn't make a difference but you can log this chassis speeds object to see if it is trying to command a rotational velocity. This could help narrow down the problem.

You could also set the custom logging callbacks of the command to log the target chassis speeds as well.

Crossle86, 2023-02-08:
Some feedback. I switched to sending chassis speeds into the drive code, logging them. They show x, y and rotation speeds. So for some reason our drive code is not responding to those rotation speeds. So strange as it works for joystick input passing the same data.

Crossle86, 2023-02-08:
So using the chassis speeds returned from PPSwerveControllerCommand, is that chassis speeds in robot oriented state or do I need to convert from fieldoriented as I do in my driving code? I tried this but no joy, but want to verify what the returned chassis speeds represent.

I am stumped. I run my driving code through the same method (accepting chassis speeds) used by PPSwerveControllerCommand and the driving works correctly but PP output does not. Rotation values present both ways. Can't see what the difference is.

Crossle86, 2023-02-08:
Update: I found the problem. It is the code handling gyro angle during sim. So under sim, it is now working. Will try on real robot soon. One thing, the rotation from PP is opposite the joystick rotation. Seems odd but I can't state that our code is doing all the X/Y/ROT absolutely correct, though it is to the best of my knowledge. For now, just inverting the rotation before feeding it to the drive code.

Crossle86, 2023-02-08:
New problem: My code follows path created from the PP App pretty well. When I try this manual path:
``
        PathPlannerTrajectory exampleTrajectory = PathPlanner.generatePath(
            new PathConstraints(MAX_WHEEL_SPEED, MAX_WHEEL_ACCEL), 
			new PathPoint(new Translation2d(startingPose.getX(), startingPose.getY()), startingPose.getRotation(), startingPose.getRotation()),
            new PathPoint(new Translation2d(1.0, 1.0), Rotation2d.fromDegrees(0), Rotation2d.fromDegrees(0)) 
        );
``
I expect robot (in sim) to move 1 meter forward and 1 meter left from blue wall. Instead it moves backwards and right and more than 1 meter in each direction. It also has a tendency to generate chassis speeds with the values  = NAN which crashes the field2d display to point I have to restart the sim.

Crossle86, 2023-02-08:
'''
        PathPlannerTrajectory exampleTrajectory = PathPlanner.generatePath(
            new PathConstraints(MAX_WHEEL_SPEED, MAX_WHEEL_ACCEL), 
			new PathPoint(new Translation2d(startingPose.getX(), startingPose.getY()), startingPose.getRotation(), startingPose.getRotation()),
            new PathPoint(new Translation2d(1.0, 1.0), Rotation2d.fromDegrees(0), Rotation2d.fromDegrees(0)) 
        );
'''

mjansen4857, 2023-02-08:
First of all, you don't want to negate the rotation coming from pathplanner. That is already in the correct conventions which is positive = counter clockwise rotation. You will want to negate your joystick value instead.

Second, the positions you give to the generate path method are not relative to each other. They are field positions. So the code you posted does not "move 1 meter forward and 1 meter left". It instead moves from the "startingPose" to (1, 1) on the field. Maybe this starting pose is (0, 0) but from your description it sounds like it might not be.

Crossle86, 2023-02-08:
Actually I already negate the JS values to match the field convention. That is why this is puzzling. Same with JS rotations. Not sure why I have to negate the rotation following path. Our drive code expects + rotation to be ccw. Still something missing. Have had some issues with sim so I think this is a sim issue.

As to the manual path, I will adjust according to what you said. I took transform to be an offset to be added to the previous pose. I haven't found a good discussion of the Translation2d and Transform2d so I don't fully understand what they are used for.

Crossle86, 2023-02-08:
Changing the transform values as suggest fixed the problem. In addition to whatever you can tell me about translations/transform, can you clarify the different between heading and rotation in the pathpoint?

Crossle86, 2023-02-08:
It seems like pose, translation2d and transform2d all refer to a point. Makes sense for pose, the other to seem to suggest relative movement from a pose. Reading the WPILib doc did not help much...

WarrenReynolds, 2023-02-08:
> Changing the transform values as suggest fixed the problem. In addition to whatever you can tell me about translations/transform, can you clarify the different between heading and rotation in the pathpoint?

Rotation only comes into play when using a holonomic drive train (like Swerve)

Heading is the direction that the centre of the robotics is travelling in at that point on the path. (As it follows the line)  Rotation is the direction the robot is facing with reference to the field at that point on the path.  This doesn't have to be the same as the direction the robot is travelling in.  

IE: The robot could be facing 45 degrees to the right, but driving a heading 45 degrees to the left.  In this case the robot front is at right angle to the direction of travel.  Pressing the play button in Pathplanner shows the robot moving along the path and gives you a visualization of the robot as it goes along the path.  The dot on the robot is the front of the robot. If you have two different holonomic rotation values for two different path points. The robot will rotate as it travels along the path.

Hope this helps.

WarrenReynolds, 2023-02-09:
> It seems like pose, translation2d and transform2d all refer to a point. Makes sense for pose, the other to seem to suggest relative movement from a pose. Reading the WPILib doc did not help much...

Pose is made of of an X and Y and Theta.  This will define the position of your robot anywhere on the field. (Provided in not on it's side or upside down)
X and Y are the Coordinates of the centre of the robot. Theta is the direction the robot is facing.

A Translation2D is just an X and Y value, no Theta, This gets a bit tricky, as deep down it a representation of a deltaX and DeltaY (IE a movement on the X Y plan) but in a lot instances this structure is used to represent an X,Y point on the field.  You can think of these points being a DeltaX and DeltaY from the position 0,0 so effectively they become just an X,Y point.  You can construct one of these with a distance and angle, but the angle bit is just the polar coordinate angle, not a pose heading.  You can also ask what the angle is and again this is just the angle of the x,y point from the 0,0 position.

The Transform2D is like a Translation2D, but it also includes a delta Theta.  This one is more often used to transform from one pose to another pose.   IE a movement in the X and Y direction with a change of the way the robot is facing (delta theta)

In the true sense, if you want to store an x,y point you probably should use the pose structure and ignore the theta, but in most of the WPILib code that wants an X,Y point they always used a Transform2D.

Another tip, when you see Transform2D() in code this usually mean a point to is initialized to x=0,y=0.

The way I see them is
Pose=x,y,theta 
Translation2D=x,y  
Tranform2D=DeltaX, DeltaY, DeltaTheta.

Did you search the API for each of these words.  This is how I got my head around it when I started playing with Swerve Path Following.
[image]

Hope this helps.

WarrenReynolds, 2023-02-09:
> Sorry, I am still trying to figure out how to post code into GH text box.

Tip
It's not the single quote symbol (ie the one next to the enter key). It's the key up in the top left corner of the key board to the left of the 1! key (ie below esc on a windows keyboard)

for single line of code start with one and finish with one all on the same line.

For a block of code start with 3 on a line by its self and then your code and then three more on a new line after your code.

Clicking on the preview tab at the top of the text entry box where you type will show if you have got it right or not.

Crossle86, 2023-02-09:
Thanks for the detailed reply. It seems that pose, translation, transform all refer to a point relative to 0,0 origin (ignoring angle). BTW, the glossary does not address these terms.

So it appears these items and paths all work with points +x,+y from 0,0 which is lower left blue side. So what about starting from the red side? Is the starting point of a red path a point offset from 0,0 blue, and paths go backwards, ie negative offsets from the starting point? I know there is a mirroring function, but that does not apply to this year. I know in the sim if I specify a starting pose on the red side which is a large x offset from 0,0 blue and create path points that are negative from each other (pose x going down as robot moves left) it appears to work. But am I understanding this correctly? Is this how it will work on the actual field?

mjansen4857, 2023-02-09:
0,0 is the right hand corner of your current alliance wall. The mirroring function is there for that reason because a path that needs to turn right on the blue alliance would need to turn left on the red alliance to do the same thing.

Crossle86, 2023-02-10:
So this is confusing. If I draw a path in  PP gui starting on the red size moving left toward blue, it starts at x=14, not 0. X value goes down as path moves toward blue. If on red and starting from 0,0 being right side of red, the path should start x=0 instead of 14. Hope this makes sense.

mjansen4857, 2023-02-10:
Draw all of your paths on the blue side of the field. Let PPLib handle transforming the paths when you are on the red alliance. 

0,0 is on the blue side in PathPlanner because it needs to be one of them. As far as your robot is concerned 0,0 should be on your side of the field. This usually doesn’t matter with PathPlanner, since most games are rotationally symmetric. But this game is not, so a path made on the blue side of the field needs to be tweaked on the red side.

Crossle86, 2023-02-10:
Ok. So PPLib takes the blue path and relocates it in relation to the red 0,0 point? I think I am getting it. So it would seem the PPLib transform of the path would not work with the field2d display under sim as that would require the field2d origin to change. Is  that correct? Sim only supports  blue paths?

I assume this also works for manually created paths (generatePath)? It appears so from looking at the code.

Crossle86, 2023-02-10:
So begs the question. How do we handle this years unusual field layout? Is PPLib able to handle the red switch? I seem to remember a mention of this issue but can't turn it up just now.

mjansen4857, 2023-02-12:
The sim field2d widget does not support changing the origin. I would suggest taking a look at AdvantageScope. It supports this and can automatically set it for you.

Manually generated paths will not be automatically transformed. It is up to you to give the generate path methods the correct coordinates relative to your alliance, or transform the path using the static methods in `PathPlannerTrajectory` before following the path.

PPLib is able to handle the red alliance mirroring if you set the `useAllianceColor` boolean to true in the constructor of the path following command or auto builder. Again, this will only automatically transform paths from the GUI.

Crossle86, 2023-02-12:
Thanks so much for all the information. I think I mostly get it but still some discomfort. Seems to work in sim but real robot testing sort of works and sort of does not. Needs tuning. Unfortunately higher priority items have to be addressed so path following will be set aside for now. Will probably be back later :-)

closed 2023-04-05 by mjansen4857