mjansen4857/pathplanner #241: PathPlannerTrajectory.transformStateForAlliance appears to be incorrect?

vfurman-gh, 2023-02-09:
Hello, 
First of all, thanks a lot for sharing the path planner code! It is a big help for the team I mentor!
We encountered an issue with the library, though. We trying to use PathPlannerTrajectory.transformStateForAlliance,  looking at the PathPlannerTrajectory.java:124, it looks like translation happens by mirroring around axis with Y=FIELD_WIDTH_METERS/2:
 `new Translation2d(state.poseMeters.getX(), FIELD_WIDTH_METERS - state.poseMeters.getY());`

I think to transform to red alliance coordinates, we should actually mirror around axis with X=8, so the code should look like:
`new Translation2d(16 - state.poseMeters.getX(), state.poseMeters.getY());`
What do you think?

vfurman-gh, 2023-02-09:
> The origin is always assumed to be the right corner of your *current
> alliance station*. This years game makes that a little more difficult
> because the field is not symmetrical in that way, thus mirroring over the Y
> axis is correct.
Ok, that starts making sense. So April tags locations need to be
transformed for alliance as well? I wonder if it would be simpler to have
origin fixed in the blue alliance corner always and have transformation
only for PP trajectory? Wdyt?

hadley31, 2023-02-09:
Yes, the `AprilTagFieldLayout` class has a `setOrigin()` method, that should be set depending on the value of `DriverStation.getAlliance()`

closed 2023-02-12 by mjansen4857