mjansen4857/pathplanner #242: Adding rotation when there should not be any

C34343, 2023-02-09:
Here at 1730 we are new to pathfinding this year, and this is the best library we have found for path finding. Currently we are trying to get it fully set up, and we are testing a simple drive forward path. Through our debugging we have found a few things. 
1. It moves how we expect with both teleop driving, and a command to drive forward, so it isn't a physical problem
2. The trajectory does not have any rotation when sampled, but after we return the SwerveModuleState array it adds rotation randomly.
3. The wheels don't rotate at a constant state, first they drive right, then turn towards forward before turning back, moving the whole time.

Here is the code we are using https://github.com/TeamDriven/RobotCode2023 
The files of interest would be the Drivetrain subsystem, the DriveForward auto, and possibly some of the other drivetrain commands or the robot container.

WarrenReynolds, 2023-02-09:
There are few things you need to troubleshoot here.

Is your odometry working properly?
IE: If you zero your pose in odometry and then in tele-op, drive straight away form yourself down the field for 3 metres, does the odometry position of the robot report to be 3m, 0m, 0 deg?
And If you drive 45 degree to the left for 1.41m without any rotation, does you odometry report 1m, 1m, 0 deg?
Also if you drive forward for 1m and rotate 90 degrees to the right as you are driving, does your odometry report 1m, 0m, -90 Deg?

The path following code is trying to correct any errors in the x coordinate, y coordinate and theta value as you drive along the path.  It samples the path, checks odometry and then depending on your PID values, corrects for any errors in finds in each axis. For the straight line case, if the robot drifts to the left in odometry (ie y>0) the PID controller will multiple this error by Kp and then apply a negative Y velocity, this is converted into swerve module states that will try to move the robot to right to get it back on path. So even though you path is "Straight" It can still cause the swerve modules to turn.

Lots of people are saying "That it works in Tele-Op, but not in auto", sometimes people aren't actually testing the a push forward on the joystick is resulting in a Positive x velocity, and push left on the joystick is resulting in a positive y velocity and a counter clockwise Twist is resulting in a Positive Theta Rate.  I thinking a lot of people aren't negating the joystick values and they just turn the robot around and say that must be the front because it goes forward I push the joystick forward.

To troubleshoot, I find it easier to return the Chassis Speed from the path follower instead of the swerve module states, than you can easily pull out the x,y,theta velocities and dump them to a dash board to see whats happening.

When we were getting all this going we dumped everything we could to the dash board and then changed the displays to graphs.  This helped us a lot.

C34343, 2023-02-10:
We checked our odometry and it is fully working. It moves with only a small difference between odometry and reality. We were able to get it to drive straight forward, but only when we set the kPYController to 0. This isn’t a good fix. We even tried making the kPYController negative, but it just changed the direction we slid in.

mjansen4857, 2023-02-10:
You aren't setting `useAllianceColor` to false and your PPLib version is not up to date so it is not set to false by default. Because of this your path is being transformed for the red alliance and you are not transforming the initial state when resetting odometry so the initial state will be off from the rest of the path. I suggest updating PPLib so that `useAllianceColor` will default to false. You may also want to use `SwerveAutoBuilder` since that will handle resetting odometry and stuff properly for you when you enable `useAllianceColor` for competitions.

chauser, 2023-02-12:
We experienced something related to this. We built modeled our code pretty directly on the examples in https://github.com/mjansen4857/pathplanner/wiki/PathPlannerLib:-Java-Usage. The various PathPlanner Commands examples there all show a SequentialCommandGroup containing an InstantCommand resetting the odometry just before a PPXXXCommand with  `useAllianceColor` set to true. The misbehavior is just as suggested in the the preceding remark -- a trajectory intended to move a short distance in the X direction instead moves a long distance in the Y direction when the alliance color is Red. When we changed `useAllianceColor` to false all was well.
Our paths thus far don't need the transformation; and before they do we'll change over to SwerveAutoBuilder for multiple reasons.

closed 2023-02-21 by C34343