mjansen4857/pathplanner #243: Path Doesn't Reverse When Reversed is Checked Under Generator Settings

cozymentor, 2023-02-10:
Path tries to drive forward no matter how the check box is set.

mjansen4857, 2023-02-10:
If you are using a holonomic drive, the reversed flag is not supposed to do anything, there is not concept of driving "forwards" or "backwards" with a holonomic path.

If you are using a differential drivetrain, the checkbox in the GUI has no effect on paths in code. You need to set the `reversed` boolean to true in the `loadPath` method.

cozymentor, 2023-02-10:
We are using differential.  That makes sense, I will give that a go.  Thank you!

closed 2023-02-10 by cozymentor

Kevin-OConnor, 2023-02-26:
I'm curious why this is the case? How is my code supposed to know whether the path is supposed to be reversed or not? The flag is stored in the JSON file so is there a structural reason why it's passed into the loadPath method rather than read from the file?

mjansen4857, 2023-02-26:
To keep paths fairly robot agnostic. The values in the json are only really for the preview. The code should know it needs to be reversed cuz you know which auto you’re running I guess. 

This really could be changed because the event markers stuff sort of kills the whole robot agnostic thing. But I could see a lot of people being confused as to why their robot drives backwards not knowing they selected a reversed check box. This also happens the other way around so I guess there’s not a great reason not to. Something to look at for future releases.

Kevin-OConnor, 2023-02-26:
Thanks, love the app overall, just stubbed our toe a bit here today and figured I'd try to understand the design. Chalk me up as an anecdote on the confused by the current behavior side. It felt like the path on the robot should match the preview in the GUI. It took us quite a while to realize that the "Reversed" checkbox wasn't affecting the behavior and from there very quickly see the overloads with the reversed flag. It was easy enough to grab the flag out of the path file myself (so certainly no urgency on our end) but just felt like something the library should be doing.

mjansen4857, 2023-02-26:
Yeah. The current biggest issue with it is that the generator settings card would need to be more visible, since it can get hidden behind the preview mode.