mjansen4857/pathplanner #246: How do I make the robot go slow in the middle of my path

EddieDL, 2023-02-11:
I want to have a path that moves X distance at one velocity then travel Y distance at a slower velocity finally travel the rest of the path going the original velocity.

I have tried to use the Vel Override functionality on waypoints, but that doesn't seem to work like I want. It basically make the robot "slow down" for the specific point. I need it to travel a specific distance much slower than the rest of the path.

I think I might be able to create stop points at the start/end of where I want to go slow and then pass in specific PathConstraints to the loadPathGroup method (3 total PathConstraints). I am going to try that, but would love some advice on the "right" way to do this.

mjansen4857, 2023-02-12:
> I think I might be able to create stop points at the start/end of where I want to go slow and then pass in specific PathConstraints to the loadPathGroup method (3 total PathConstraints). I am going to try that, but would love some advice on the "right" way to do this.

This would be the way I'd recommend. The velocity override thing needs a bit of a rework to apply to more than just the waypoint. You should be able to set a velocity override at your stop points if you want to transition into the slower segments without stopping.

EddieDL, 2023-02-13:
Sounds good. That does work for me. Thanks for the response.

closed 2023-02-13 by EddieDL