mjansen4857/pathplanner #252: Rotating without Translating

richardthedou, 2023-02-16:
Let's say that I want to rotate a swerve robot without translating. In holonomic mode, if I set the start point and the endpoint to be coincident or very close to each other and only change the holonomic rotation (let's say by 180 degrees), the rotation will be very fast. The only solution I have is to use a very low acceleration/velocity constraint (0.01m/s^2 for example). However, it seems counter-intuitive. Is there any other way I could rotate a robot by 180 degrees in 1 second without translating for example?

mjansen4857, 2023-02-16:
Use a turn in place command. A path planning program really isn't for turn in place stuff.

Also, if you have a swerve drivetrain there shouldn't really be any situations where you would want to turn in place since you can rotate while following the path.

richardthedou, 2023-02-17:
Oh thanks, I think that's reasonable.

closed 2023-02-17 by richardthedou