mjansen4857/pathplanner #257: Feature Request: make PathPlanner.State fully public or add getter functions

randomstring, 2023-02-18:
Making these class variables private prevents wiring functions that iterate over a trajectory. For instance, if you want to write a function that transforms a trajectory in the same way that `transformStateForAlliance` does, but with a different translation that does not require a coordinate system change. 

https://github.com/mjansen4857/pathplanner/blob/c6ee3221c1b6e61522e5da74a8cad03e4b45668c/pathplannerlib/src/main/java/com/pathplanner/lib/PathPlannerTrajectory.java#L536-L537

randomstring, 2023-02-18:
I found this reference on CD where you say those two State values are "ornamental" but there's still the problem of the constructor being protected.

https://www.chiefdelphi.com/t/pathplanner-2023-beta/416548/206

mjansen4857, 2023-02-21:
Going to change this issue to encompass some other things that need to be made public

jasondaming, 2023-03-01:
Adding the [add events to a on the fly generated path](https://www.chiefdelphi.com/t/pathplanner-2023-beta/416548/281?u=jdaming) use case to this issue.

closed 2023-03-03 by mjansen4857