mjansen4857/pathplanner #262: C++ AutoBuilder Event Map usage causing crashes when using event map commands more than once

r4stered, 2023-02-25:
Hello,

I am trying to use the events feature of pathplanner with C++ and am running into some a crash whenever I try and use the same event command twice. All of the events commands are returned from a CommandFactory in each relevant subsystem. 

Here is what I have in my Autos.cpp file:

`  std::unordered_map<std::string, std::shared_ptr<frc2::Command>> map{
    {"MoveOutOfStartingConfig", std::move(m_armSub->GoToPose([]{ return ArmPose::OutOfStartingConfig(); }).Unwrap())},
    {"MoveArmToHighPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::ScoreConeHigh(); }).Unwrap())},
    {"MoveArmToMidPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::ScoreConeMid(); }).Unwrap())},
    {"MoveArmToStowedPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::StowedConfig(); }).Unwrap())},
    {"PoopPiece", std::move(m_intakeSub->PoopGamePiece(1_s).Unwrap())},
    {"IntakeObject", std::move(m_intakeSub->IntakeGamePiece(1_s).Unwrap())},
    {"MoveArmToGroundIntake", std::move(m_armSub->GoToPose([]{ return ArmPose::GroundIntakeFar(); }).Unwrap())}
  };

  eventMap = std::make_unique<std::unordered_map<std::string, std::shared_ptr<frc2::Command>>>(std::move(map));
  
  autoBuilder = std::make_unique<pathplanner::SwerveAutoBuilder>(
    [this] {
      return m_driveSub->GetRobotPose();
    },
    [this](frc::Pose2d resetToHere) {
      m_driveSub->swerveDrivebase.ResetPose(resetToHere);
    },
    m_driveSub->swerveDrivebase.GetKinematics(),
    pathplanner::PIDConstants(str::swerve_drive_consts::GLOBAL_POSE_TRANS_KP, 0, str::swerve_drive_consts::GLOBAL_POSE_TRANS_KD),
    pathplanner::PIDConstants(str::swerve_drive_consts::GLOBAL_POSE_ROT_KP, 0, str::swerve_drive_consts::GLOBAL_POSE_ROT_KD),
    [this](std::array<frc::SwerveModuleState, 4> states) {
      m_driveSub->swerveDrivebase.DirectSetModuleStates(states);
    },
    *eventMap.get(),
    std::initializer_list<frc2::Subsystem*>({m_driveSub}),
    true
  );`
  
  Am I doing something wrong, or should I create separate command entries for each event on my list?

My code in full for reference is here: https://github.com/frc2053/Robot2023/

mjansen4857, 2023-02-25:
What exactly do you mean by "more than once"?

Also, if you have an error print out, that would be helpful to determine what is up. Are you running the latest PPLib version? There was recently a fix that could have something to do with this.

WarrenReynolds, 2023-02-26:
From the commit history on the PathplannerLib.json it looks like PP Lib was version 2023.4.1 up until 19 Feb and currently at 2023.4.2 currently. 
@r4stered Is there a chance that the command you are running twice hasn't finished it's first instance when it's being called a second time? 

I don't know much about shared and unique pointers, but could the issue be in this code
```
std::unordered_map<std::string, std::shared_ptr<frc2::Command>> map{
    {"MoveOutOfStartingConfig", std::move(m_armSub->GoToPose([]{ return ArmPose::OutOfStartingConfig(); }).Unwrap())},
    {"MoveArmToHighPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::ScoreConeHigh(); }).Unwrap())},
    {"MoveArmToMidPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::ScoreConeMid(); }).Unwrap())},
    {"MoveArmToStowedPosition", std::move(m_armSub->GoToPose([]{ return ArmPose::StowedConfig(); }).Unwrap())},
    {"PoopPiece", std::move(m_intakeSub->PoopGamePiece(1_s).Unwrap())},
    {"IntakeObject", std::move(m_intakeSub->IntakeGamePiece(1_s).Unwrap())},
    {"MoveArmToGroundIntake", std::move(m_armSub->GoToPose([]{ return ArmPose::GroundIntakeFar(); }).Unwrap())}
  };
```
I think .Unwrap() returns a unique pointer and the map is supposed to have shared pointers.  I thought the idea of unique pointer was so that only one copy of them could exist and if this map is being passed around in code I think the pointers would need to be shared pointers.

As a test you could throw in some std::make_shared and see if that helps.

Also, you could take a few variables out of the equation in your troubleshooting and try some real basic commands like

`("marker1",std::make_shared<frc2::PrintCommand>("Passed Marker 1"))`
Then see if you can use marker1 more than once.

r4stered, 2023-02-26:
Here is some more tests I ran. Here is my pathplanner path I am running:

[image]
[image]
[image]

This worked:

```
  std::unordered_map<std::string, std::shared_ptr<frc2::Command>> map{
    {"IntakeObject", std::make_shared<frc2::PrintCommand>("Passed marked one!\n")}
  };
```

This also worked:

```
Autos.cpp
std::unordered_map<std::string, std::shared_ptr<frc2::Command>> map{
  {"IntakeObject", m_intakeSub->IntakeGamePiece(1_s).Unwrap()}
};

IntakeSubsystem.cpp
frc2::CommandPtr IntakeSubsystem::IntakeGamePiece(units::second_t howLongToSpin) {
  return frc2::PrintCommand("Passed Marked One!\n").ToPtr();
}
```

But when trying to chain multiple commands in the Command Factory it crashes:

```
  return frc2::cmd::Sequence(
    frc2::cmd::Print("Passed Marked Start!"),
    frc2::cmd::Wait(1_s),
    frc2::cmd::Print("Passed Marked End!")
  );
```

So I was originally wrong that multiple usages causes this, but there is still an issue where if you try to do a sequence of commands it doesn't work.

Here is the output:

```
Passed Marked Start!
Passed Marked End!
Assertion failed: idx < size(), file C:\Users\willi\dev\allwpilib\wpiutil\src\main\native\thirdparty\llvm\include\wpi/SmallVector.h, line 276
Warning at frc::Watchdog::Impl::Main: Warning: Watchdog not fed within 0.020000s
Error at frc::IterativeRobotBase::PrintLoopOverrunMessage: Error: Loop time of 0.020000s overrun
CommandScheduler loop time overrun.
```

[image]

r4stered, 2023-02-26:
```
frc2::CommandPtr IntakeSubsystem::IntakeGamePiece(units::second_t howLongToSpin) {
  return frc2::cmd::Sequence(
    frc2::cmd::Print("Passed Marked Start!")
  );
}
```

It still manages to run the PrintCommand, so it seems like when it tries to clean up it crashes.

I tried with pathplanner 2023.4.2 as well as 2023.4.1 and that didn't make a difference.

 Maybe something with the sequence being destructed is messing it up?

Starlight220, 2023-02-26:
Is there a message printed on crash?
CommandPtr has some checks for use-after-move.

r4stered, 2023-02-26:
No messages on crash, just the out of index assert print. I got rid of the moves and that didn't make a difference. I think when you implicitly convert a unique_ptr to shared_ptr it does that anyway.

WarrenReynolds, 2023-02-26:
@mjansen4857 I don't any command that has member variables is going to work in wrappedEventCommand, because all you are doing is copying the init, exec, isfinish and end as you are building the wrapped event.  
And I think a command sequence has an internal vector of commands as a member variable, that it runs using its own internal logic. The wrapped event won't be able to see these will it?

Starlight220, 2023-02-27:
The out-of-index assert print is that `SequentialCommandGroup.Execute()` is trying to run with an out-of-bounds index. This means that `Execute` was called either before `Initialize` or after `IsFinished` returned true.

---
> And I think a command sequence has an internal vector of commands as a member variable, that it runs using its own internal logic.

I don't think that's the problem here; `SequentialCommandGroup` has safeguards for case of empty vector.

WarrenReynolds, 2023-02-27:
The code I'm talking about is this code, where Michael creates a new local variable of type frc2::FunctionalCommand called wrapped.  He then sets up the Initialize, Execute, End, IsFinished and the requirements to be the same as the command that is passed to wrappedEventCommand.  I think the idea here is so that a command in the event map can be used over and over again.  But this doesn't account for commands/Command Group that maintain some internal state information like the sequential and parallel command groups do.  Basically any command/ command group that has a member variables that get manipulated will break using this method.  Only one instance of the command exists in the event map, so there is only one set of member variables/ state information that can be manipulated. If the command group in the event map is used a 2nd time, I'm guessing the internal logic of the command group is in an unpredictable state.  I'll go looking through the source for the sequential command group and report back 
```
frc2::CommandPtr BaseAutoBuilder::wrappedEventCommand(
		std::shared_ptr<frc2::Command> command) {
	frc2::FunctionalCommand wrapped([command]() {
		command->Initialize();
	},
	[command]() {
		command->Execute();
	},
	[command](bool interrupted) {
		command->End(interrupted);
	},
	[command]() {
		return command->IsFinished();
	}
	);
	wrapped.AddRequirements(command->GetRequirements());

	return std::move(wrapped).ToPtr();
}
```

WarrenReynolds, 2023-02-27:
After looking at the source, I'm none the wiser.  @Starlight220, you are correct, the sequential command group init makes sure that the index is reset and valid.

On further thought, could it be in the implementation of FollowPathWithEvents.cpp.  In here, Michael has written his own logic to execute a group of commands.  I'm struggling to follow the code at the moment, but I'm thinking this is the next spot to look.  Someone needs to copy PP Lib Locally and add a bunch on fmt::prints into here to see exactly when the init and execute are being called.

r4stered, 2023-02-27:
I think I see what is happening. I added a bunch of prints to FollowPathWithEvents, and it seems like tracking the running commands isn't working correctly. Because even though the command pair is set to not run anymore, the check for this still thinks the command is running:

```
		if (!runningCommand.second) {
			fmt::print("We are not running a command yet!\n");
			continue;
		}
```

Here is a log:

```
Done with running command loop!
eventCommand = Sequence Wrapper in Subsystemn Factory
Initializing event command with name: Sequence Wrapper in Subsystemn Factory
Passed Marked Start!
Added event command with name: Sequence Wrapper in Subsystemn Factory to current commands
Done with follow path with events execute loop!
Running command name = Sequence Wrapper in Subsystemn Factory
Before executing running command
After executing running command
Running command is now finished
Ended running command
Done with running command loop!
Done with follow path with events execute loop!
Running command name = Sequence Wrapper in Subsystemn Factory
Before executing running command
Assertion failed: idx < size(), file C:\Users\willi\dev\allwpilib\wpiutil\src\main\native\thirdparty\llvm\include\wpi/SmallVector.h, line 276
Warning at frc::Watchdog::Impl::Main: Warning: Watchdog not fed within 0.020000s
Error at frc::IterativeRobotBase::PrintLoopOverrunMessage: Error: Loop time of 0.020000s overrun
CommandScheduler loop time overrun.
```

Here is the modified version of FollowPathWithEvent with prints
```
#include "pathplanner/lib/commands/FollowPathWithEvents.h"

using namespace pathplanner;

FollowPathWithEvents::FollowPathWithEvents(
		std::unique_ptr<frc2::Command> &&pathFollowingCommand,
		std::vector<PathPlannerTrajectory::EventMarker> pathMarkers,
		std::unordered_map<std::string, std::shared_ptr<frc2::Command>> eventMap) {
	m_pathFollowingCommand = std::move(pathFollowingCommand);
	m_pathMarkers = pathMarkers;
	m_eventMap = eventMap;

	this->AddRequirements(m_pathFollowingCommand->GetRequirements());
	for (PathPlannerTrajectory::EventMarker marker : m_pathMarkers) {
		for (std::string name : marker.names) {
			if (m_eventMap.find(name) != m_eventMap.end()) {
				auto reqs = m_eventMap[name]->GetRequirements();

				if (!frc2::RequirementsDisjoint(m_pathFollowingCommand.get(),
						m_eventMap[name].get())) {
					throw FRC_MakeError(frc::err::CommandIllegalUse,
							"Events that are triggered during path following cannot require the drive subsystem");
				}

				this->AddRequirements(reqs);
			}
		}
	}
}

void FollowPathWithEvents::Initialize() {
	m_isFinished = false;

	m_currentCommands.clear();
	fmt::print("m_currentCommands cleared!\n");

	m_unpassedMarkers.clear();
	m_unpassedMarkers.insert(m_unpassedMarkers.end(), m_pathMarkers.begin(),
			m_pathMarkers.end());

	m_timer.Reset();
	m_timer.Start();

	m_pathFollowingCommand->Initialize();
}

void FollowPathWithEvents::Execute() {
	m_pathFollowingCommand->Execute();
	if (m_pathFollowingCommand->IsFinished()) {
		m_pathFollowingCommand->End(false);
		m_isFinished = true;
	}

	for (std::pair<std::shared_ptr<frc2::Command>, bool> runningCommand : m_currentCommands) {
		fmt::print("Running command name = {}\n", runningCommand.first->GetName());
		if (!runningCommand.second) {
			fmt::print("We are not running a command yet!\n");
			continue;
		}

		fmt::print("Before executing running command\n");
		runningCommand.first->Execute();
		fmt::print("After executing running command\n");
		if (runningCommand.first->IsFinished()) {
			fmt::print("Running command is now finished\n");
			runningCommand.first->End(false);
			fmt::print("Ended running command\n");
			runningCommand.second = false;
		}
	}
	fmt::print("Done with running command loop!\n");

	units::second_t currentTime = m_timer.Get();
	if (m_unpassedMarkers.size() > 0
			&& currentTime >= m_unpassedMarkers[0].time) {
		PathPlannerTrajectory::EventMarker marker = m_unpassedMarkers[0];
		m_unpassedMarkers.pop_front();

		for (std::string name : marker.names) {
			if (m_eventMap.find(name) != m_eventMap.end()) {
				auto eventCommand = m_eventMap[name];
				fmt::print("eventCommand = {}\n", eventCommand->GetName());

				for (std::pair<std::shared_ptr<frc2::Command>, bool> runningCommand : m_currentCommands) {
					if (!runningCommand.second) {
						fmt::print("{} command is not running!\n", runningCommand.first->GetName());
						continue;
					}

					if (!frc2::RequirementsDisjoint(runningCommand.first.get(),
							eventCommand.get())) {
						fmt::print("{} command has conflicting reqs with {}!\n", runningCommand.first->GetName(), eventCommand.get()->GetName());
						fmt::print("Ending {}...\n", runningCommand.first->GetName());
						runningCommand.first->End(true);
						runningCommand.second = false;
					}
				}

				fmt::print("Initializing event command with name: {}\n", eventCommand->GetName());
				eventCommand->Initialize();
				m_currentCommands.emplace_back(eventCommand, true);
				fmt::print("Added event command with name: {} to current commands\n", eventCommand->GetName());
			}
		}
	}
	fmt::print("Done with follow path with events execute loop!\n");
}

void FollowPathWithEvents::End(bool interrupted) {
	if (interrupted) {
		m_pathFollowingCommand->End(true);
	}
	
	fmt::print("Currently running commands on end:\n");
	for (std::pair<std::shared_ptr<frc2::Command>, bool> runningCommand : m_currentCommands) {
		fmt::print("{}\n", runningCommand.first->GetName());
		if (runningCommand.second) {
			fmt::print("that command was running! Now ending it!\n");
			runningCommand.first->End(true);
		}
	}
}

bool FollowPathWithEvents::IsFinished() {
	return m_isFinished;
}
```

r4stered, 2023-02-27:
Ah I figured out the actual bug. We need to get the running commands by reference in the for each loop, or else it makes a copy. So runningCommand.second = false is modifying the copy, not the one we actually want to change.

WarrenReynolds, 2023-02-27:
Great Work @r4stered

WarrenReynolds, 2023-03-05:
Does the PR that was done to fix this problem automatically flow through to the source code that is linked with the path planner Vendor dependence entry or does that source only get updated when a new release is released?

Sorry for such a dumb question, I don't fully understand the whole marvinURL, gradle, vendor dep stuff and how that pulls in code.

The reason I ask is because we are using the path planner vendor dep and we are still having the same problem with the code crashing when a command sequence was used in event map.  
This is what the PathplannerLib.json file look like in our project.
```
{
    "fileName": "PathplannerLib.json",
    "name": "PathplannerLib",
    "version": "2023.4.2",
    "uuid": "1b42324f-17c6-4875-8e77-1c312bc8c786",
    "mavenUrls": [
        "https://3015rangerrobotics.github.io/pathplannerlib/repo"
    ],
    "jsonUrl": "https://3015rangerrobotics.github.io/pathplannerlib/PathplannerLib.json",
    "javaDependencies": [
        {
            "groupId": "com.pathplanner.lib",
            "artifactId": "PathplannerLib-java",
            "version": "2023.4.2"
        }
    ],
    "jniDependencies": [],
    "cppDependencies": [
        {
            "groupId": "com.pathplanner.lib",
            "artifactId": "PathplannerLib-cpp",
            "version": "2023.4.2",
            "libName": "PathplannerLib",
            "headerClassifier": "headers",
            "sharedLibrary": false,
            "skipInvalidPlatforms": true,
            "binaryPlatforms": [
                "windowsx86-64",
                "linuxx86-64",
                "osxuniversal",
                "linuxathena"
            ]
        }
    ]
}
```

Is there any way to see the .cpp source code that is being linked in with a vendor dep entry.  I noticed that if you F12 on the #include pathplanner/lib/pathplanner.h line it will show you the header, but Alt+O does switch to the source.
Thanks in advance for any help.

mjansen4857, 2023-03-05:
Looks like you’re not up to date. Current version is 2023.4.3

WarrenReynolds, 2023-03-06:
Thanks.
This fixed it.