mjansen4857/pathplanner #267: Using the path duplicate function does not copy non-pathpoint markers.

WAaron-500, 2023-02-27:
Path to be copied:

[image]

Path when copied:

[image]

WAaron-500, 2023-02-27:
Also, cannot add new markers to a duplicated path as well.

WAaron-500, 2023-02-28:
Forgot to add, when running the copied path, the markers also don't exist (commands attached to the markers don't run)

mjansen4857, 2023-02-28:
Yeah noticed this a while back but its hard to get time to work on it. I believe it lets you add markers if you close and reopen the app.

WAaron-500, 2023-02-28:
Yeah, that's what I was thinking do you know where all the GUI stuff is located, I can help fix it if I know which file to look at... not very good at programming, but Ill try my best

PR #291 (2023-03-27): Markers update
Closes #267 and maybe #288 

Also seems to fix not being able to add markers at least for me.
closed 2023-03-27 by mjansen4857