mjansen4857/pathplanner #271: Stop point being skipped

jonahsnider, 2023-03-04:
Expected behavior:
https://user-images.githubusercontent.com/7608555/222873960-2f831025-7733-431c-b70f-f46853a8b830.mov
[image]

Actual behavior:
https://user-images.githubusercontent.com/7608555/222873965-4c4d237b-10f7-4014-b687-7e4f107b882c.mov
[image]

Path:
[bug.path](https://github.com/mjansen4857/pathplanner/files/10887382/bug.path.txt)

Trajectory `.toString()`:
[trajectory.txt](https://github.com/mjansen4857/pathplanner/files/10887414/trajectory.txt)

The actual behavior is using simulated robot code, but we first noticed the issue on a real robot.
If you're unable to reproduce this bug locally, I can try creating a reproduction repo, as I am unable to share my team's full source code until after the season.

The simulation is super barebones, we just "teleport" the robot wherever Pathplanner is asking us to go:
[image]

We load the path in a standard manner, and follow it using `SwerveAutoBuilder#followPath()`:
[image]
[image]
[image]

mjansen4857, 2023-03-04:
Stop points are for path groups. Loading it as a single path will not split the path into a group.

jonahsnider, 2023-03-04:
Good catch! Our path works as desired once we load & follow it as a group. Do you think there's anything that could be changed in PathPlannerLib to make it harder to make this kind of mistake? I feel like this is probably something that happens to lots of users.

closed 2023-03-04 by jonahsnider