mjansen4857/pathplanner #273: Velocity Override Is Almost Useless

Nyxiad, 2023-03-04:
There are certain sections where we want a much lower max speed; however, the velocity override is only for one point. I would like for the entire path between two consecutive points to inherit the higher of the velocity overrides from those points.

mjansen4857, 2023-03-05:
Velocity override is mainly just used to set a start/end velocity for the path. What you’re describing can be done with stop points and path groups. When loading a path group you can set vel/accel for each segment. If you use a stop point to create the path group just set a velocity override at the point so it will have a non zero velocity going through the point. 

Next years version should have this stuff be a bit more clear and make more sense. But the velocity override being changed like you describe would be redundant and kind of defeats the purpose of what it is actually for. If I want the path to end with a non zero velocity, making the whole final segment that velocity would slow things down massively compared to just right at the end.

Nyxiad, 2023-03-06:
Ah, I'm sorry. I forgot to add that this is only for the AutoBuilder. That makes sense, thanks!

closed 2023-03-06 by Nyxiad

mjansen4857, 2023-03-06:
Even with the auto builder you can still change the constraints of each segment. You just need to add more to the loadPathGroup function

Nyxiad, 2023-03-06:
Yep, I just finished adding that to our code. Thanks again!