mjansen4857/pathplanner #274: Add Angular Velocity/Acceleration Constraints

Nyxiad, 2023-03-04:
The feedforward values seem to be way too high which causes our robot to rotate much faster than we are comfortable with.

mjansen4857, 2023-03-05:
Not currently possible without a full generation rewrite. Going to add this to the 2024 milestone but no guarantees on that.

WarrenReynolds, 2023-03-06:
I think the fast spin often happens because people change the holonomic rotation by 180 degrees on two points that are reasonably close together.  It appears as if the time allocated to get between those points is based on the distance between them and the Max Vel and Max Acc values, as the distance is short the time is short, As a work around, could you just put really low velocity overrides in for the points on either end of segment to give it more time to complete the spin?  

A future feature that would be nice is to have a max velocity settable for the segment between two points.

Nyxiad, 2023-03-08:
#273 "A future feature that would be nice is to have a max velocity settable for the segment between two points."

PR #317 (2023-08-08): 2024 Reworks
Majority of the reworks for the 2024 release.

Fixes #316, closes #311, closes #304, fixes #301, closes #289, closes #285, closes #274, closes #260, fixes #226, fixes #210, closes #163, closes #138, closes #126

Supersedes #300 

TODO:
- [X] GUI reworks
- [X] Java PPLib reworks
- [X] C++ PPLib reworks (missing pathfinding for now)
- [X] Test Java PPLib in robot project
- [x] Documentation
- [x] Finish GUI unit tests
- [x] Test (and probably fix) C++ PPLib in robot project