mjansen4857/pathplanner #280: Strange stop event behavior

Eggy-T, 2023-03-07:
I'm using pathplanner do do autonomous for my robotics team, and I can't get stop events to work. The robot will wait at the stop event's location and follow the wait instructions I put in, however no events will fire off at stop events.
Normal event markers do work, and my event map does have everything that I'm trying to run in it.
My code is available at https://github.com/TeamHollywood7211/2023-Conduit
My roborio is up to date w its formatting and imaging.

This could totally just be a flaw with my code, and I might be posting this problem in the wrong location, but I tried the FRC discord and we couldn't figure it out, so I came here.

I love the program by the way, it's made creating an autonomous SO much easier.

mjansen4857, 2023-03-08:
First of all, I see you are using an event map in RobotContainter, but I don’t see it instantiated anywhere. Maybe you accidentally removed it before committing?

Also, if I had to take a guess at what is happening here, I believe it’s because you are creating your autos before you actually populate your event map. Because you’re just creating them outside of the constructor, these calls to the full auto method will be made before you add the commands to the event map in the robot container constructor. 

Normal event markers don’t use the event map until you run the command, but stop events use the event map when the command is created. So, move the creation of your auto modes to the end of your robot container constructor, and this should fix this issue.

Eggy-T, 2023-03-08:
This is exactly what it was, I ended up figuring it out today just before I read this. Initially, I had the event map in my constants without anything in it, and then i populated the event map in the robot container constructor. I moved everything into the robot container and defined it all at the same time and it worked flawlessly.

Thank you so much!

closed 2023-03-08 by Eggy-T