mjansen4857/pathplanner #286: PathPlanner Not Connecting to Server on MacOS

burakdemirelli, 2023-03-20:
Having tested on different Macbooks, all connected to the robot wirelessly, Pathplanner could not connect to the real-time diagnostics tool in the app. Trying it on a Windows machine resolved it for now. If there is more information needed, I can collect it and provide it.

Ardusa, 2023-04-28:
Was this ever resolved?

burakdemirelli, 2023-04-30:
> Sorry if I already asked this on CD but is this for the app store version or the version downloaded from github?

Sorry, I did not see this before. This was seen on the version downloaded from the app store.

mjansen4857, 2023-10-05:
This should be fixed in the main branch already with the switch to using network tables so I will close this unless the issue pops back up again.

closed 2023-10-05 by mjansen4857