mjansen4857/pathplanner #293: Option to detect run-away divergent path following and stop execution

clrozeboom, 2023-03-28:
We had an issue during one of our matches where we had a SparkMax fail us (lost power) and the position estimation updates were causing wildly inaccurate pose results. In thinking about how to handle this, one option that occurred to me was that I'd like the path follower to be a little more aware of run away behavior and have the option to stop following the path. Poses should fall in the range of 0 to 27 feet in the Y direction and 0 to 54 feet in the X, possibly shorter depending on the alliance white line in a given game. They also should be converging on the expected pose within a limit that could be specified, say no more than 0.5 meter of divergence before stopping the auto. Thoughts? I did something similar to this on the side of the drive-function, (restraining pose to the field or stop motors in auto) but the question of how to detect excess trajectory path divergence even when inside the field remains. The ability to write a custom supplied function might be enough.

mjansen4857, 2023-04-05:
I don't think this is something that should really be handled in pathplanner, but in your own codebase. If stopping a path when you go far off course is something you want then you can implement that fairly easily with something like `pathFollowingCommand.until(() -> runaway condition)`

The condition that would actually trigger this is really specific to the team/robot as well as how it is handled. Some teams wouldn't want it to end at all. For instance, at one of our competitions, we ran a 2 + balance auto mode, but our alliance partner's auto went off course and rammed into us. Because the path continued to run, it was able to get back on course and complete the balance even though it was knocked pretty far off course.

Basically, what I think you're looking for is hardware fault detection that would stop an auto, not detecting something wrong with path following that would stop it. There's pretty simple ways to detect this already. Our code periodically checks (~4Hz) the firmware version of our spark max or talon, if an error is returned then you have no connection with the motor controller. If this motor controller is one of your drive related motors, stop the path following.

clrozeboom, 2023-04-05:
Thanks, those are some interesting options.

closed 2023-07-31 by mjansen4857