mjansen4857/pathplanner #297: Issues with using Alliance color mirroring

GaCo1629, 2023-04-03:
I'm a new user so be gentle..

I'm trying to use the mirroring function to save path development time, but I'm seeing odd behavior.  When I run Blue, all is OK,

Red is the problem.

I'm replicating the code found here https://github.com/mjansen4857/pathplanner/wiki/PathPlannerLib:-Java-Usage for PPSwerveControllerCommand.

I am displaying the Swerve Pose on my dashbaord, and when I run on the Red alliance side, the path seems OK, but the starting location that is loaded by the isFirstPath option still reflects the Blue starting position.   So, the travel directions have been flipped, but its still thinks it's starting at the same field location

This might be OK in some situations, BUT I'm also doing state estimation using the AprilTags, so once the robot sees a tag, it starts correcting the field location to match the actual X/Y location, which is way different than where PathPlanner wants it to be.

Am I doing something wrong here, or this mirroring option not suitable for full field coordinates.

GaCo1629, 2023-04-03:
After further investigation, I see that the Mirroring performed by the PPSwerveController isn't compatible with AprilTag State estimation as the Red Alliance field coordinates are not compatible.

So the logical thing to me would be to either add a switch to the loadPath method which translates the waypoint and ControlPoint X&Y values to the other end of the field, or to just have a flipToRed() method which can morf a path from blue to red doing the same thing.

Anyone tackled this task?

PatribotsProgramming, 2023-04-05:
#263

PatribotsProgramming, 2023-04-05:
Here is a snip bit of our code that would act as the "flipToRed()" method that you mentioned.
`
PathPlannerState state = (PathPlannerState) _pathTraj.sample(elapsedTime);
PathPlannerState mirroredState = new PathPlannerState();

// System.out.print("Desired position " + state.poseMeters.getTranslation() + "\n Swerve pose: " + swerve.getPose().getTranslation());

// Create a new pathplannerstate based on the mirrored state's position
// and taking the mirrored state's rotation and adding 180 degrees
if ((DriverStation.getAlliance() == DriverStation.Alliance.Red) && DriverStation.isAutonomous()) {

  // state.poseMeters = new Pose2d(
  //   (AlignmentConstants.FIELD_WIDTH_METERS - state.poseMeters.getTranslation().getX()),
  //   state.poseMeters.getTranslation().getY(),
  //   state.poseMeters.getRotation().unaryMinus().plus(Rotation2d.fromDegrees(Math.PI)));

  // state.holonomicRotation = state.poseMeters.getRotation().plus(Rotation2d.fromRadians(Math.PI)).unaryMinus();

  mirroredState.timeSeconds = state.timeSeconds;
  mirroredState.velocityMetersPerSecond = state.velocityMetersPerSecond;
  mirroredState.accelerationMetersPerSecondSq = state.accelerationMetersPerSecondSq;
  mirroredState.poseMeters = new Pose2d(
      (AlignmentConstants.FIELD_WIDTH_METERS - state.poseMeters.getTranslation().getX()),
      state.poseMeters.getTranslation().getY(),
      state.poseMeters.getRotation().unaryMinus().plus(Rotation2d.fromDegrees(Math.PI)));
  mirroredState.curvatureRadPerMeter = state.curvatureRadPerMeter;
  mirroredState.holonomicRotation = state.holonomicRotation.plus(Rotation2d.fromRadians(Math.PI)).unaryMinus();
  mirroredState.angularVelocityRadPerSec = state.angularVelocityRadPerSec;
  mirroredState.holonomicAngularVelocityRadPerSec = state.holonomicAngularVelocityRadPerSec;

  // System.out.println(" To " + mirroredState.poseMeters.getTranslation());

}
// Use elapsedTime as a refrence for where we NEED to be
// Then, sample the position and rotation for that time,
// And calculate the ChassisSpeeds required to get there
ChassisSpeeds _speeds = HDC.calculate(
    swerve.getPose(),
    // Pass in the alliance to flip on the Y if on red alliance
    ((DriverStation.getAlliance() == DriverStation.Alliance.Red) && DriverStation.isAutonomous()) ? mirroredState : state,//(DriverStation.isAutonomous()) ? mirroredState : state,
    ((DriverStation.getAlliance() == DriverStation.Alliance.Red) && DriverStation.isAutonomous()) ? mirroredState.holonomicRotation : state.holonomicRotation);//(DriverStation.isAutonomous()) ? (mirroredState.poseMeters.getRotation()) : state.holonomicRotation);

// Set the states for the motor using calculated values above
// It is important to note that fieldRelative is false,
// but calculations make it, so it is true i.e. rotation is independent
// (This is seen 6-5 lines above)
swerve.drive(
    _speeds.vyMetersPerSecond,
    _speeds.vxMetersPerSecond,
    _speeds.omegaRadiansPerSecond, false, false);`

You can find it at our [github](https://github.com/Patribots4738/ChargedUp2023/blob/main/src/main/java/auto/SwerveTrajectory.java#L162)

gearsincorg, 2023-04-13:
> #263

Hey.  Thanks a lot.  That's exactly what I needed, but really didn't have the expertise to code it that nicely.

Phil.

closed 2023-07-31 by mjansen4857