mjansen4857/pathplanner #321: 2024.0.0.24 version, GUI Simulation doesn't always follow path

WarrenReynolds, 2023-08-14:
I'm not sure if this is by design (ie some sort of optimization/path smoothing feature) or it's a bug.  But I've noticed that the center of the robot doesn't follow the path all the time.  It works perfectly for the example path but when you move some of the control points there are times when the robot (ie graphical  representation of the robot) completely leaves the path.  I noticed this when trying to get the robot to go around the sharp corner of the charge station.  

BTW: Absolutely love the new version.

mjansen4857, 2023-08-14:
Yeah this is how the new path following works. It uses the pure pursuit algorithm and accelerates/decelerates on the fly instead of the whole motion profile being generated ahead of time. So, the paths you create aren't really the exact path that the robot will follow any more, they're sort of a target path you want the robot to try following.

Going to simplify it a bit so I don't explain the whole algorithm, but the lower the robot's acceleration, the further ahead in the path it needs to target in order to start following a curve without completely overshooting. But, because its targeting so far ahead it can start making a curve before actually reaching the curve in the path, causing undershooting.

I've tried to minimize this as much as possible with a few different things but the path following just can't really assume as much as the previous version since it all works based on the robot's actual state, not what the generation pre-determined it should be at a certain point in time. I still think that its a good trade-off to have a bit of overshoot/undershoot in some cases in order to gain the benefit of being resistant to error caused by collision or getting stuck on things, being able to correct odometry while path following without it being super jerky, and the ability to follow paths with discontinuities (such as paths generated with A* or the new autos that can have multiple paths without the guarantee that their start and end points perfectly line up). 

You *should* be able to tune out most overshoot/undershoot by playing around with global constraints and constraint zones at problem areas. But, if you find any really egregious examples of over/undershooting that you can't seem to get rid of at reasonable speeds then feel free to send over the path file so I can see if I can find a way to prevent it from happening.

WarrenReynolds, 2023-08-14:
I'm torn with these new changes. I sort of liked the previous moving setpoint, X, Y and Theta controller model. I can see the benefits in smoothing and pure pursuit, but I also feel that if you have a control point in your path the robot should be doings its hardest to track through these control points and not cutting corners in the name of pure pursuit.  I'll have to play with the constraints to see how it plays out.  

I think the path that is drawn on the GUI should be the path that the robot "should" follow if turned correctly.  I know it would take a big re-write, but could you update the GUI to display the pure pursuit path and not the Beizer path.  This would mean that the path would actually change as you change constraints (either global, zonally, or control point constraints).  Or maybe have an option that show a dotted path line that shows the projected pure pursuit path with the given constraints.  All of the code to generate the pure pursuit path is already there for the animation so all you would need to do is plot a center dot as you draw the animation of the robot following the path, and then clear these dots before restarting the animation.

Another option would be to some sort of optional dynamic adjustment of the look ahead distance that changes as you approach and leave a control points.  This would force the robot to track harder to the control points.

Thanks again for making your path planning program publicly available.

WarrenReynolds, 2023-08-14:
Side note.  I thought I'd have a play with the right hand side of the GUI (Autos section) to see how the effect of joining paths with discontinuous headings plays out in regards to the new pursuit stuff. 

I can add an Auto by clicking the + symbol, but as soon as I try an edit it, it crashes the program.  When you come back to the program the autos section is empty again.

mjansen4857, 2023-08-15:
Fixing that crash in #323 

So, there is not any actual intentional smoothing going on. The path you see is the one the robot "should" follow. It just might not be actually capable of doing so perfectly in some situations. This was also the case with the old algorithm as well, difference being that the old controller would violate the path constraints in order to minimize error. It might command the robot to go much faster than the max velocity to catch up to where it should be, or apply a greater than max acceleration in curves in order to stay on track. New algorithm actually enforces these constraints at all times (except for final deceleration to make sure it actually stops at the right spot).

Showing the simulated path would also be a bit misleading, since this is just a perfect world simulation. Robot may perform differently if its not actually capable of following with the constraints given, if the modules are not tuned properly, etc.

WarrenReynolds, 2023-08-15:
> Or maybe have an option that show a dotted path line that shows the projected pure pursuit path with the given constraints

OR an option that leaves every plot of the animated robot rectangle drawn on the screen.  This would effectively leave behind a big fat path that "should" show where the outer edges of the robot "should" have travelled while using pure pursuit.  If you see this fat semi transparent path is going to collide with a field element than you know that you have to adjust your path control points.

Only ideas.

mjansen4857, 2023-08-15:
It depends. The simulation is really only valid if you are starting to follow the path from a stop and at exactly the start point, which isn't necessarily the case when you actually follow the path. You can follow any path starting from any position with any velocity and it will work. The auto simulation is much more useful than the path simulation because of this. Still not a guarantee it will actually do what it shows, as with the old path preview system. But, I can see how the whole path of the simulation would be useful over just the robot outline at a given point in time. Will look into a way of doing that.

WarrenReynolds, 2023-08-15:
OK, that makes a lot more sense now.  I was only thinking from an autos perspective (ie starting at the start of the path).   I didn't realize that you now have the ability to approach a path from anywhere on the field at anytime in the match and effectively merge onto it using pure pursuit.  That's going to be a pretty cool feature.  

How does that even work? Does it find the closest perpendicular point on the path and head for that and then depending on how far from the path you are, the look ahead target starts to move along the path when you get inside the lookahead distance?  

You are really going to need the ability to customize the look ahead distance for tight spaces to force the robot to hug the path (like past the edges of the charge station for the season just gone).  And I suppose that it what the customizable zonally velocity and acceleration constraints are all about.

Can't wait to see this in action.  Hey another nice feature could be virtual walls.  IE: If a look ahead projection is sending you on a collision course with a wall, field element or penalty zone, have the pure pursuit following algorithm steer you around it.  And I suppose that is what  A* is all about.

Looking forward to seeing the 3015 robot in action in 2024.  This is going to be sick.

mjansen4857, 2023-08-15:
It basically chooses a point on the path X distance away from the robot. If there is no point within the lookahead distance, it will target the start of the path. This distance is determined based on how fast the robot is going, as well as its max acceleration. It needs to be further away for lower accelerations or it will massively undershoot. Trade-off being you get some undershoot to avoid a lot of overshoot.

Decreasing the lookahead distance doesn't necessarily make the robot follow the curve closer. If it is too low the robot won't start accelerating in the direction of the curve until it is too late, which causes the overshoot. Basically, the robot tries to drive towards the lookahead point. So, you can imagine that if this point is really close to the robot, when it starts to follow a curve the needed heading change for the robot would be far more than it could actually handle with limited acceleration, causing the overshoot.

mjansen4857, 2023-08-19:
#327 lets you show the center path of the simulation so you can compare it to the original bezier curve (new editor settings dropdown in the path/auto tree). Also made a tweak to the lookahead distance calculation that should make it track a bit closer. Still though, there are situations where it won't be perfect, so some manual tuning for your specific robot will be needed if you want it perfect. This will be possible before any beta release.

closed 2023-08-19 by mjansen4857