mjansen4857/pathplanner #325: Pathplanner closes immmediately

AndromedaHelix, 2023-08-17:
Whenever I open PathPlanner, the app closes instantly. It doesn't matter if I use the one from Github, AppStore, or directly build it from source. Does anyone knows how to fix this?

The only "insight" I have is that when I build it from source, it opened normally, but as soon as I opened a robot project, it closed. It is important to note that the file I chose was the deploy file, by error. Other times I have used Pathplanner, I have correctly opened everything

AndromedaHelix, 2023-08-17:
I am using Mac M1 by the way

mjansen4857, 2023-08-18:
Can you run `defaults delete com.mjansen4857.pathplanner` in the terminal then try again.

WarrenReynolds, 2023-08-18:
It could also be a corruption in one of your existing path files in the deploy folder for that project. This fits with the symptoms you are describing above. 

Try taking all of the files out of the deploy folder and restarting the app.  If that works, close PP and copy one path back in at a time, opening and closing PP each time.

mjansen4857, 2023-08-18:
If you're building from the current main branch that shouldn't be an issue anymore but for older versions, yeah.

AndromedaHelix, 2023-08-18:
Yeah I did all of the above and it worked. Thanks a lot!

closed 2023-08-18 by AndromedaHelix