mjansen4857/pathplanner #334: Replace pure pursuit with trajectories + path replanning

mjansen4857, 2023-08-26:
Following a path with pure pursuit introduces a lot of error. Rework the path following to use generated trajectories again and replan a path if necessary.

WarrenReynolds, 2023-08-26:
I like this approach.  Will we get the option back to export excel files of the generated trajectories from the app?  That was a nice feature that we used in our FLL team last year.  It was only open loop control (ie no telemetry feedback correction), but with a bit of adjusting of feed forwards and heading correction using gyro feedback you could get an FLL robot to track spine based paths across an FLL mat really nicely.

I'm not sure how you are planning to implement the "re plan if necessary" but it would be really nice if there was some way of handing control of the drivebase over to a command for last minute alignment for picking up game pieces or scoring game pieces.  I look forward to seeing how you are going to implement this.

mjansen4857, 2023-08-26:
> Will we get the option back to export excel files of the generated trajectories from the app?

I don't think this is something I want to support anymore. The 2023 version of the app is still capable of doing so and will remain available to download. But the new version would require some changes specifically to be able to support it, because the concept of not having a set starting state for every path will remain. So, you can't generate a trajectory ahead of time without making assumptions that might not be true, or adding to the UI specifically to do so. PathPlanner is an FRC tool first and foremost and I don't think it makes sense for anyone to use the pre-generated files over the vendorlib anymore so it doesn't make sense to put work into supporting that going forward.

> I'm not sure how you are planning to implement the "re plan if necessary"

Path following command will create a new path + generate a trajectory if your starting position is not near the starting position of the original path, or if you are starting with a non-negligible linear velocity. The new path will re-join with the pre-planned one as soon as is reasonable.

>  it would be really nice if there was some way of handing control of the drivebase over to a command for last minute alignment

You could do this by just interrupting the path following command. Or, if you're doing it in an auto, just have the path end when you want the auto alignment to take over and give it a non-zero end velocity if you want a smooth transition, then start the auto alignment command.

For something like taking over rotation to help align to a game piece while path following, what we did last year was just edit the output chassis speeds' rotation component to target the game piece instead of directly outputting the path following command's output to the modules.

closed 2023-08-29 by mjansen4857

WarrenReynolds, 2023-09-24:
> The 2023 version of the app is still capable of doing so and will remain available to download

How do I download the 2023 version from github?