mjansen4857/pathplanner #341: Add setting to adjust snap distance

hadley31, 2023-08-29:
The snap distance currently feels too large, and is hard to fine tune without shifting the value manually in the text box, or disabling the setting and re-enabling it. An alternative or additional option is to bind snapping to a held hotkey (CTRL) like most game engines do. I'd personally like to see both features. I can look into adding them myself if I find the time!

mjansen4857, 2023-08-30:
Not sure if a setting to change the distance is really worth it, since once that distance is dialed in to something that feels right (its currently 0.25, 0.1 is probably better) no one would want to increase it, and anyone that would want to decrease it further would probably just want it off at that point.

Also, instead of holding ctrl to enable snapping, I think it would be better if holding ctrl just did the opposite of whatever the current snapping setting is. So, even if you have it enabled you have a hotkey for fine control whenever you want it.

hadley31, 2023-08-30:
That's a fair point! Those suggestions sound good to me!

closed 2023-08-30 by mjansen4857