mjansen4857/pathplanner #357: Can not delete rotational targets in latest release

WarrenReynolds, 2023-09-20:
I was just testing the latest version with the adjustable rotational target.  Nice feature.  Unfortunately the delete icon on rotational target is not working.  It is currently functioning as an expand/hide instead of delete.
The bin icon circled in white is what i'm referring to.
On a side note is it possible to add the functionality of the green circle with the number in it..  IE have a number that shows the number of items under a collapsed menu option.  For example the path shown in the screenshot has two constraint zones, but unless you know they are there you would have no idea until you expand out that section.  That feature could apply to all collapsed menu items (if it is possible to do that) 

The other option would be to add a show/hide icon to each section in the GUI.  If you wanted to see the zones in the GUI, turn on the eye next to constraint zones.  If you wanted to hide the event marker points because the GUI is too cluttered, you could turn off their visibility.  I'm thinking like the Fusion 360 browser or Adobe Illustrator Layers.

This could be further expanded to include an enable/disable icon in the GUI. If something is disabled, it appear in the GUI, but would not be included in the path generation or the commands wouldn't be added to the Auto group, if all of the commands were disabled.

I feel this would be a really nice feature to be able switch in and out different things without the need to delete them entirely from the GUI when testing.  You could add two more boolean fields to the JSON data for each item in a path, visible and enabled.   Maybe this could be a feature request for 2025.
  
[image]

mjansen4857, 2023-09-20:
The delete button is working fine on my end. Does it work if you make the tree wider? You have it really small so there may be some overlap issues causing the button press to go to the expansion tile instead of the delete button.

WarrenReynolds, 2023-09-20:
Works full screen, but not when its windowed.

WarrenReynolds, 2023-09-20:
Thoughts on the feature requests?

mjansen4857, 2023-09-20:
[image]
You need to drag this thing to make the tree wider, not the whole window size. The current minimum size causes a bunch of issues with the UI in the tree so I'll need to increase it a bit.

The features sound nice but at some point adding all that stuff would make the tree UI really cluttered. I can try to play around with it eventually.

WarrenReynolds, 2023-09-20:
Agreed, If I had to prioritise, I'd say the circle with the number of items under the collapsed group before the enable/disable, show/hide functionality. Happy for this to be moved out of issues and into discussion if you are considering these options or just close and delete if you want.

closed 2023-10-06 by mjansen4857

WarrenReynolds, 2023-10-08:
Any chance you can make the numbers a little bit brighter?  It could be my monitor, but they are very hard to see.

[image]