mjansen4857/pathplanner #360: Unable to launch v2023.4.2 on Mac

LTeixeira4909, 2023-10-03:
```
flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: │ FileSystemException: Creation failed, path = '/deploy' (OS Error: Read-only file system, errno = 30)
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ #0   _Directory.createSync (dart:io/directory_impl.dart:129)
flutter: │ #1   _Directory.createSync (dart:io/directory_impl.dart:124)
flutter: │ #2   _HomePageState._loadPaths (package:pathplanner/pages/home_page.dart:532)
flutter: │ #3   _HomePageState.initState.<anonymous closure>.<anonymous closure> (package:pathplanner/pages/home_page.dart:112)
flutter: │ #4   State.setState (package:flutter/src/widgets/framework.dart:1133)
flutter: │ #5   _HomePageState.initState.<anonymous closure> (package:pathplanner/pages/home_page.dart:109)
flutter: │ #6   <asynchronous suspension>
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ ⛔ Dart Error
flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

LTeixeira4909, 2023-10-03:
It seems to happen because I mistakenly choose a directory that is unwriteable. Now that directory seems to be cached somewhere

mjansen4857, 2023-10-03:
`defaults delete com.mjansen4857.pathplanner` in terminal. Closing since this bug should be fixed in main branch.

closed 2023-10-03 by mjansen4857