mjansen4857/pathplanner #444: Robot Code Crashing When Single Path Following Code is Deployed

SecretGenius22, 2023-11-01:
As it says in the title every time I try to push my robot code to the robot the code on the rio crashes and I have to power cycle the rio for it to start working again. I am using the Pathplanner 2024 Beta version 4. Using auto builder to run an auto path works, but trying to run a single path as it says in the wiki is what breaks it. A screenshot with the code causing the problem is below, and the github repository is here: https://github.com/team223/223-Robot-Project-Pre-Season-2024
[image]

mjansen4857, 2023-11-01:
Haven't really looked at it yet but I'm going to assume it is because you're trying to "turn in place". PathPlanner is not meant for doing this. Use a separate command for turning in place and only use PathPlanner for following paths.

SecretGenius22, 2023-11-01:
Thanks for the response, I fixed the name, this path follows a straight line, and uses holonomic movement to turn 90 degrees as it drives. 
[image]

SecretGenius22, 2023-11-01:
I don't think I changed anything besides the name of the path, but the code is able to deploy now without crashing and the path runs.

closed 2023-11-01 by SecretGenius22