mjansen4857/pathplanner #450: Pathplanner auto paths rotate as fast as possible towards a rotation target.

elliot1234567, 2023-11-03:
So when my team tested autonomous routines with pathplanner, we discovered that the robot will rotate at its maximum angular velocity in order to reach the rotation target as fast as possible. This has caused us some issues with tracking position accurately, as the swerve modules can't keep up with the set voltage. In the old version of pathplanner, this wasn't an issue as the robot would rotate smoothly between waypoints, using the full distance to make the rotation. Would it be possible to add a feature where the user can decide whether or not they wish to use smooth rotation between waypoints (or rotation targets) on a per path basis?

Thanks!!

mjansen4857, 2023-11-05:
In the old version, it would rotate smoothly, but it was never constrained. So, it could be told to rotate much faster than possible depending on the distance it had to complete the rotation. If it is going too fast now, reduce the rotation constraints to slow it down. Also, if you want to make a rotation last over a longer distance, you can use a constraints zone to slow down the rotation over just that distance.

Rotation targets are just fundamentally different from how rotation worked in previous versions. Its not a "reach this rotation at this point" thing anymore its "try to rotate to this rotation until this point" if that makes sense. Rotation is handled at run time now instead of during generation to handle things like being able to start from anywhere w/ any rotation, set a rotation override, and reduce rotation speed to avoid saturating modules. With all of this it gets more difficult to get rotation to work in the style of previous years.

I might try to look into it for the future, but not sure when.

Ramtech59Sensei, 2023-11-13:
> So when my team tested autonomous routines with pathplanner, we discovered that the robot will rotate at its maximum angular velocity in order to reach the rotation target as fast as possible. This [...]
Elliot if you add a constrain that only change the rotation speed and acceleration in the section of the path that you want the robot to rotate slow, then it should rotate slowly 
If you plan to use it with a phat finding then it may not finish turning but it should work other way 
Correct me if I'm wrong "mjansen4857"

closed 2023-11-19 by mjansen4857