mjansen4857/pathplanner #451: PathPlannerAuto not working at all

0gottogo0, 2023-11-03:
`Unhandled exception: java.lang.IncompatibleClassChangeError: class com.pathplanner.lib.commands.PathPlannerAuto has interface edu.wpi.first.wpilibj2.command.Command as super class
Error at java.base/java.lang.ClassLoader.defineClass1(Native Method): Unhandled exception: java.lang.IncompatibleClassChangeError: class com.pathplanner.lib.commands.PathPlannerAuto has interface edu.wpi.first.wpilibj2.command.Command as super class
at java.base/java.lang.ClassLoader.defineClass1(Native Method)
at java.base/java.lang.ClassLoader.defineClass(Unknown Source)
at java.base/java.security.SecureClassLoader.defineClass(Unknown Source)
at java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(Unknown Source)`
this error appears in the riolog every time I try and run an auto. My code is on github at this link: https://github.com/0gottogo0/Swerve-drive-test

mjansen4857, 2023-11-05:
You need to use the 2024 WPILib beta. PPLib 2024 is not compatible with WPILib 2023.

closed 2023-11-05 by mjansen4857

0gottogo0, 2023-11-07:
Oh