mjansen4857/pathplanner #459: PathPlanner sometimes after deploy has paths that do not follow the written path at all.

PGtheVRguy, 2023-11-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11]
 - GUI Version: [e.g. 2024.1.1]
 - PPLib Version: [e.g. 2024.1.1]
 - PPLib Language: [e.g. Java]

**Additional context**
Add any other context about the problem here.

mjansen4857, 2023-11-07:
Based on the video you linked, this looks like a gyro failure. Notice how it continues to drive backwards robot-relative as it rotates when leaving the community. This indicates that the gyro is probably not working here since the path following commands attempt to drive field relative. Driving field relative without a working gyro essentially just ends up being robot relative.

mjansen4857, 2023-12-07:
Closing since its probably a gyro issue

closed 2023-12-07 by mjansen4857