mjansen4857/pathplanner #463: Pathfind to Path only works when robot is already moving in the direction

cwoodard1121, 2023-11-11:
When using pathfindToPose from the autobuilder class, if I try and run it while the robot is not moving, it will just freeze. When I run it moving in the general direction of the path it will work though.

**To Reproduce**

`PathPlannerPath path = PathPlannerPath.fromPathFile("Cone1");

PathConstraints constraints = new PathConstraints(3.0, 3, 540, 720);

Command pathfindingCommand = AutoBuilder.pathfindThenFollowPath(path, constraints, 3.0);

new Trigger(() -> OI.DR.getStartButton()).onTrue(pathfindingCommand);` 

**Expected behavior**
I expected it to move towards the path regardless of the robot's velocity

**Versions: (please complete the following information):**
 - OS: Windows 10
 - GUI Version: 2024.0.0-beta2
 - PPLib Version: 2024.0.0-beta-4
 - PPLib Language: java

**Additional context**
Also when starting the pathfind to pose it seems to carry on in the moving direction for longer than necessary before getting on the path, and goes outside the navgrid not sure if this is intentional.

cwoodard1121, 2023-11-12:
running followPathWithEvents seems to do the same thing as pathFindThenFollowPath and works.

mjansen4857, 2023-11-12:
Can you give a little more info here? Like videos or sample code or something that causes this. As well as the "Cone1" path and your navgrid.json if you edited it. I am unable to reproduce.

cwoodard1121, 2023-11-13:
changed a little bit, using a seperate class on button click but shouldn't be an issue.
navgrid is not edited.

PathConstraints constraints = new PathConstraints(3.0, 3, 540, 720);
PathPlannerPath path = PathPlannerPath.fromPathFile("BlueCleansideCone1");
Command pathfindingCommandGrid = AutoBuilder.pathfindThenFollowPath(path,constraints,0);

CommandScheduler.getInstance().schedule(pathfindingCommandGrid);

[BlueCleansideCone1.zip](https://github.com/mjansen4857/pathplanner/files/13330135/BlueCleansideCone1.zip)

cwoodard1121, 2023-11-13:
demo, https://youtu.be/4T3ZOQX5lAM

seems to work if my current velocity will intercept the path at some point but otherwise nothing it sorta just freezes.

mjansen4857, 2023-11-13:
Yeah I still can't reproduce at all, even with the path you sent. Can you verify that you are running at least PPLib beta 4? If so, can you send me your auto builder config code and the implementations of the methods used to configure it. Better yet, if you send a link to your project I can just run that.

A few things I noticed that may or may not have an effect:
* The path you're running is really weird here. Its going across the whole field which is what the pathfinding is supposed to do. Also, pathfindTheFollowPath is really meant for when the pathfinding will be travelling towards a path that will continue in the same general direction. But, I still can't reproduce even with the path you sent.
* The angular velocity constraints are in radians, not degrees. So, you're giving it insanely fast constraints.
* The control lengths you have set are way too small. Setting them this small causes a lot of problems, which is why it is limited to a minimum of 0.1 since beta 4.

cwoodard1121, 2023-11-13:
[pathplannertesting.zip](https://github.com/mjansen4857/pathplanner/files/13336521/pathplannertesting.zip)
Tried changing all of that, didn't work. 
click the left stick to run the pathfinding command, hit A first to set the trajectory.

mjansen4857, 2023-11-14:
Found the problem. Seems like your version of LocalADStarAK was edited at some point to change the instances of `!Logger.hasReplaySource()` to `Logger.hasReplaySource()`. So this means that the pathfinder will only ever generate paths during replay, so it won't work at any other time.

Replacing your copy with the one [here](https://gist.github.com/mjansen4857/a8024b55eb427184dbd10ae8923bd57d) fixed your issue for me.

cwoodard1121, 2023-11-14:
thank you! no idea how that happened

closed 2023-11-14 by cwoodard1121