mjansen4857/pathplanner #464: Add reversals, stop points, and the ability to have 2 control vectors go in separate directions

Gold872, 2023-11-11:
Reversal and stop points were essential last year for being able to quickly change direction or intake something as the robot needed to be still

mjansen4857, 2023-11-21:
Was looking into this more, and it would break the current implementation of path replanning in some situations. This will probably be pushed to next year when I can get around to refactoring some path stuff to work better with it. 

This exact functionality basically already exists through the auto system anyways. Doing it that way works better with modularity as well. If the robot is going to stop, it might as well be split into separate paths to allow for other options for leaving or arriving at the point where it stops. The shared waypoint feature will make this easier.

Gold872, 2023-11-21:
This doesn't necessarily have to exist for path replanning. And if stop points or reversals don't exist, then it's essentially useless unless if you're using non-holominic mode. In the original Pathplanner that was the most significant feature (in my opinion) that separated it from pathweaver, as you weren't forced to go in one direction.

mjansen4857, 2023-11-21:
> This doesn't necessarily have to exist for path replanning.

What? The shape of the path is essential to how path replanning works.

> And if stop points or reversals don't exist

Like I mentioned they do exist through the auto system. You're thinking of paths like how they were in the 2023 version, which is fundamentally different from 2024. A path last year would be able to define an entire auto mode, that is why the reversals and stop points existed. This is now what autos are for in 2024. A path now defines a singular movement. The lead in to a reversal and the exit from a reversal are separate paths. There is no reason for stop points to exist within paths, as you would just stop in the middle of a path, do nothing, then carry on. A "stop point" is now the transition between one path and another in an auto, where you can run other commands. Reversals could make more sense, but it still just makes more sense to do this through separate paths and make use of the modular auto system. For example, have one path go pick up a game piece. Then, you have two options that either go back to score that game piece or go to the charging station to balance. Now, you can reuse the same path to actually pick up the game piece for both options. If you just used reversals within paths, each path would need its own version of the part that picks up the game piece, which is not shared between the two different auto modes.

Gold872, 2023-11-21:
> What? The shape of the path is essential to how path replanning works.

I thought you were referring to the on the fly path regeneration, I mixed them up.

Splitting the paths up in order to reverse or have a stop point would work, but it makes managing all of the paths together a huge pain and it involves constantly switching back and forth between 2 paths. The shared waypoints will most likely help with this though.

mjansen4857, 2023-11-21:
> involves constantly switching back and forth between 2 paths

This happens without using paths modularly as well. If you want to change the first piece pickup for a 3 piece and a 2+balance auto, you have to go change 2 different paths if using 2023 reversals instead of just one path in the 2024 modular system.

It could potentially be more work if you need to change the position of the transition between paths, but like you mentioned, shared waypoint positions will solve this.