mjansen4857/pathplanner #468: crash involving control length

cwoodard1121, 2023-11-13:
**Describe the bug**
pathplanner crashes after I change the previous control length to 0, then back to anything.

**To Reproduce**
1. Make a path
2. Make it have a path that naturally in the end point has "previous control length" of > 0
3. after it's set to something, type in 0 and hit enter
4. change it back to something through text and hit enter
after that it should crash

**Expected behavior**
it to not crash

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11] windows 10
 - GUI Version: [e.g. 2024.1.1] latest

mjansen4857, 2023-11-13:
This was already fixed in beta 4

closed 2023-11-13 by mjansen4857

cwoodard1121, 2023-11-13:
oh my bad my shortcut was still set to 3 thanks