mjansen4857/pathplanner #471: Posibility to change navigation grid on the fly

Ramtech59Sensei, 2023-11-13:
**Is your feature request related to a problem? Please describe.**
I don't see in the wiki any place where say "load navigation grid"
since a navigation grid for Auto can be completely different to Teleop grid. 
Can we load different grids ? 

**Describe the solution you'd like**
implement option to load the grid in the code 

**Additional context**
Example
a)  we don't want the robot to find paths than cross the line , so we create and obstacle
b)  we want the robot to path find only on 1/2 of the auto field section to prevent robot Collision

mjansen4857, 2023-11-13:
I'd recommend against using pathfinding in auto in the first place when pre-planned paths can handle what you need. The pathfinding will only pathfind across the midline if you tell it to go to a position across the midline. Same thing with the 1/2 the field example. As long as the path you expect it to follow is the shortest path, it will do that. But again, this is just limiting the pathfinding to such a degree that it just ends up being a really complicated way to generate pre-planned paths. You'd be better off just doing that from the beginning. So, I don't really see the utility in this.

Ramtech59Sensei, 2023-11-13:
ok understand

closed 2023-11-16 by mjansen4857