mjansen4857/pathplanner #473: Path planner UI not oppaning

samduckett, 2023-11-16:
**Describe the bug**
I have no idea why, but Path Planner just decided not to open it on my computer. It was working for a day, and I made some paths. Something happened, and now it will not open. I have tried reinstalling it, installing older versions, and building it Flutter. don't know if this is a problem with path planner or my computer, but I thought I would let you know. I have also tried running it from terminal

**Expected behavior**
I would expect that path planner would open when clicking on the Exe file.

**Versions: (please complete the following information):**
 - OS: Windows 11 Home 22H2
 - GUI Version:  v2024.0.0 Beta 5
 - PPLib Version: 2024.0.0-beta-5.1
 - PPLib Language: Java

**Additional context**
Add any other context about the problem here.

mjansen4857, 2023-11-16:
If you tried building it in flutter did you try running it in debug mode `flutter run`? That should give you error output.

samduckett, 2023-11-16:
this is in the log.txt

┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ RangeError (index): Invalid value: Valid value range is empty: 0
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ #0   _StringBase.[] (dart:core-patch/string_patch.dart:258)
│ #1   parseString (package:function_tree/src/interpreter.dart:43)
│ #2   parseString.binaryOperation (package:function_tree/src/interpreter.dart:155)
│ #3   parseString (package:function_tree/src/interpreter.dart:169)
│ #4   new SingleVariableFunction (package:function_tree/src/trees.dart:105)
│ #5   FunctionTreeStringMethods.toSingleVariableFunction (package:function_tree/src/extensions.dart:22)
│ #6   FunctionTreeStringMethods.interpret (package:function_tree/src/extensions.dart:29)
│ #7   NumberTextField._onSubmitted (package:pathplanner/widgets/number_text_field.dart:73)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Flutter Error
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

mjansen4857, 2023-11-16:
That error really shouldn't be possible? Anyways, if deleting that folder fixed it then that should mean something with your settings got messed up somehow. I'm fine with it for now unless it happens again and you know what caused it.

closed 2023-11-16 by samduckett