mjansen4857/pathplanner #480: Account for module speed desaturation when generating trajectories

Gold872, 2023-11-18:
I was testing out PathPlanner last week, and we encountered an issue where our path following would be way off whenever we were trying to turn and rotate, but it would work fine if we either removed the rotation, or heavily slowed down our speed. The most probable reason for this is that the module states that are being calculated by the trajectory are being desaturated, causing the robot to not be able to move at the desired speeds, and the PID controllers are trying to correct for it when it's impossible to do.

This issue could be worked around by adding a bunch of different constraints at different points, but that takes quite a while to do, and also doesn't solve the issue, it just adds a temporary work around that isn't fully reliable.

The only practical solution for this is to have options in the app for the max module velocity and also the wheel base + track width dimensions in either the app or the library. I could see this being extremely difficult to implement since it has to account for all of the speed desaturation in the trajectory generation and the driving time calculation.

mjansen4857, 2023-11-18:
This already exists to some extent with the runtime rotation rate limiting. If its not doing enough, either play around with the max module speed number provided in HolonomicPathFollowerConfig, or use constraint zones or just generally slower rotation constraints. Significant changes to the generation algorithm are not going to be looked at for 2024 at this point, I don't want to destabilize and or break things this close to kickoff. Further use of proper module constraints is something I'm already planning on looking into for next season.

A different option for handling rotation that will spread it out over the whole distance to the rotation target instead of going as fast as possible may still be added since it won't need any significant changes. This would also help with this in most cases.

mjansen4857, 2023-11-19:
#482 added the option to choose between rotation strategies mentioned above

closed 2023-11-19 by Gold872

WarrenReynolds, 2023-11-20:
Instead of rewriting the algorithm to be able to generate paths that don't saturate (Would be very hard to implement), maybe a warning/info box that states, the current path and rotational setting will generate module speeds with a max module velocity of x.y meters per second (Not as hard to implement).  Then teams could just see this warning and adjust their settings to give themselves some room.  

Obviously you can't account for the PID controllers getting behind when following a path and trying to catch up causing them to exceed the max velocity, But if a warning/ informational box said the max module speed for this path would be say 4.8 mps and the teams max speed was 5 mps, they could make a call that saturation is likely to happen.  Then they could tweak their paths/ setting to reduce the "calculated" max module speed.