mjansen4857/pathplanner #493: Logging callback for path ending

Gold872, 2023-11-27:
There isn't an easy way of knowing when to clear the trajectory from a Field2d object without either adding something at the end of a command, which wouldn't work if the command is interrupted (if time runs out).

A good solution for this would be to have another callback method for whenever the auto command is done following a trajectory, which will get called at the end of the auto command, even if it is interrupted.

mjansen4857, 2023-11-28:
I think a better solution here is just to call the active path callback with an empty list when the path following commands end. This avoids needing to add a separate callback.

Gold872, 2023-11-28:
That's easy enough to do, if you want I could make a simple PR for it

mjansen4857, 2023-11-28:
Up to you. I’ll probably be able to get to it tomorrow if not.

closed 2023-11-29 by mjansen4857