mjansen4857/pathplanner #502: Unable to create new named command inside of Auto Tab

m10653, 2023-12-04:
**Describe the bug**
It does not appear to be possible to add a new named tab within the Auto tab. It Fails to save and resets to empty.

**To Reproduce**
1. Create a Path
2. Create a Auto Containing the Path
3. Attempt to add a command after the path within the auto with a new name. (IE testing)
4. Back out of the auto menu
5. Go back to auto menu - value is not saved.

If this same thing is done with in the path tab when attempting to generate an event marker it works as expected and creates a new named command, that then can be selected.
**Expected behavior**
You should be able to create a new named command within the auto menu like you can within the Path tab. 

**Screenshots**
[image]

**Versions: (please complete the following information):**
 - OS: Windows 10
 - GUI Version:v2024.0.0-beta-5
 - PPLib Version:  Latest beta
 - PPLib Language: Java

mjansen4857, 2023-12-04:
Are you hitting enter after typing in the new name? This is a requirement for those dropdown fields that I can't work around

m10653, 2023-12-04:
Edit: that appears to be the issue Yikes. Must have hit enter on the path one without noticing.  Is there anyway to give notice to the user that enter must be hit for that field?

mjansen4857, 2023-12-07:
Added a note to the wiki. Something in the UI would clutter things up too much

closed 2023-12-07 by mjansen4857