mjansen4857/pathplanner #516: Auto turn off hot reload

scarmain, 2023-12-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
The docs say don't turn on hot reload at a competition.  Could you make this feature auto disable via DriverStation.isFMSAttached() (at least the robot listening to it?

**Additional context**
I just don't know the concern is, just that the driver station running PathPlanner will change the command?

hadley31, 2023-12-16:
I think what you're looking for already exists, you just need to do it yourself in robotInit.
https://github.com/mjansen4857/pathplanner/blob/1261fefab649b41235ca74275f597ac711ded57f/pathplannerlib/src/main/java/com/pathplanner/lib/util/PPLibTelemetry.java#L40-L42

```java
if (DriverStation.isFMSAttached()) {
  PPLibTelemetry.enableCompetitionMode();
}
```

mjansen4857, 2023-12-16:
That doesn't really solve the problem. The problem is that when hot reload is on you could accidentally change a path without deploying, then that changed path gets run in a match. Just enabling competition mode when the robot is connected to the FMS doesn't necessarily fix that, since you could still break things in the pit. You really just want to add the `PPLibTelemetry.enableCompetitionMode();` to robotInit when you go to a competition.

scarmain, 2023-12-16:
Sorry, didn't know the framework that well

closed 2023-12-16 by scarmain