mjansen4857/pathplanner #517: Import AutoBuilder Causes ValueError | Python 3.11

escime, 2023-12-18:
**Describe the bug**
When running any code with AutoBuilder imported, the following error is produced:
`ValueError: mutable default <class 'wpimath.geometry._geometry.Translation2d'> for field positionMeters is not allowed: use default_factory`

**To Reproduce**
Steps to reproduce the behavior:
Import AutoBuilder using the following code:
`from pathplannerlib.auto import AutoBuilder`

**Expected behavior**
Code should run with this import statement included as per [example code on this page.](https://github.com/mjansen4857/pathplanner/wiki/Python-Example:-Build-an-Auto)

**Screenshots**
Here's the full stacktrace I'm getting with the error.
[image]

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: Not Applicable
 - PPLib Version: 2024.0.0b6.post1
 - PPLib Language: Python 3.11

**Additional context**
Sorry if this is something I'm doing wrong, but I'm not exactly a practiced Python developer and as far as I can tell from the error it's nothing my code is doing.

mjansen4857, 2023-12-18:
This seems to be a problem with python 3.11+. I've been doing dev and testing with 3.10 and that works fine. I'll update the project to be compatible with 3.11

mjansen4857, 2023-12-18:
Was a pretty minor fix. I'll push 2024.0.0b6.post2 when #518 gets merged which should fix this for you

PR #518 (2023-12-18): make trajectory State class compatible with Python 3.11+
Fixes #517
closed 2023-12-18 by mjansen4857