mjansen4857/pathplanner #519: On the fly path is deviating into a constrained zone

Gold872, 2023-12-19:
I have a simple program that starts in the center of the community (in front of the charge station), and it pathfinds to a small trajectory that leads the robot right up to the double substation. However, for some odd reason it is going into the other alliance's loading zone.

Here is a gif of what is happening
[image]

This is how my navgrid is set up
[image]

This is using the latest version of WPILib 2024 beta (beta-4), and the pathplanner version that came along with beta 4.

As for the path following configuration, I am using the default replanning behavior, and with a relatively low max velocity and acceleration, being 4.0 m/s and 2.5 m/s^2

mjansen4857, 2023-12-19:
Are you setting dynamic obstacles in your code?

Gold872, 2023-12-22:
Yes, but they aren't anywhere near the path (it's an uninitialized field object)

mjansen4857, 2023-12-22:
It will still cause the path to be recalculated anytime dynamic obstacles are set. If you comment out wherever you are setting them it should prevent this. There is currently an issue where the robot will replan after a new pathfinding path comes in when doing curves like this.

Dynamic obstacles still has quite a few issues so its still not something officially supported. I will try to look into issues like this if they are an easy fix, but if its pretty difficult to find it will be much lower priority than everything else in the library. Although, at this point, I will probably just change dynamic obstacles to only cause a recalculation if a new obstacle would block the current path. This would mean it won't react to a better path opening back up, but should solve a lot of these random issues.

mjansen4857, 2023-12-25:
Give pplib beta 6.1 a try and see if it helps at all. I also noticed in my testing it seems to happen most while rotating. You might want to increase your rotation delay distance to do the rotation over a straight path.

mjansen4857, 2023-12-27:
Going to close this with #529 since that will change dynamic obstacles to only trigger the path to be recalculated if the obstacles interfere with the current path, and the requirements to trigger replanning were relaxed a bit in a previous PR.

closed 2023-12-27 by mjansen4857