mjansen4857/pathplanner #524: Option to force a Rotation Target to be hit

Gold872, 2023-12-25:
Right now there are 2 options for how to reach a rotation target, reach as fast as possible, and to not reach as fast as possible (not sure if there's a name for it). However, if you have a rotation target that even rotating at the maximum allowed rotation speed can't reach, there's no way to have it reach without manually adding more constraint zones, which can be quite annoying to do and time consuming.

This leads to my idea for a 3rd rotation target option: force hit rotation target. What this does is if the constraints does not allow the rotation target to be hit, it will slow down the translational speed so that it will hit the target. While this can be done by manually adding a constraint zone, it will not update if the angular velocity and acceleration constraints are met, and it also requires a lot of fine tuning.

I could see a reason for not adding this right away, as there is not much time before kickoff, and the actual implementation could get a bit complicated, but if it's possible to add then it would be a great addition, and if not then it could be added in the 2025 version.

mjansen4857, 2023-12-25:
This is very far out of the scope of PathPlanner for this season. It is a very difficult problem to solve, getting the robot to slow down to allow enough time for rotation. I tried before the first beta, but there really isn't a good way to do it that I know of besides turning it into an optimization problem.

Gold872, 2023-12-25:
In that case I'll stick to using constraints zones, which isn't ideal but it works which is what really matters

closed 2023-12-25 by Gold872