mjansen4857/pathplanner #527: Add PPLib offline installer download

Gold872, 2023-12-26:
**Is your feature request related to a problem? Please describe.**
Not an active problem, but one that could exist. At competitions my team and several others like to have an offline installer of all of the libraries they use in case if something goes wrong with gradle cache, however path planner doesn't have an offline installer available.

**Describe the solution you'd like**
Offer an offline installer for PPlib

mjansen4857, 2023-12-27:
The artifacts for release builds are available from the workflow [here](https://github.com/mjansen4857/pathplanner/actions/workflows/build-pplib-release.yaml). These can be placed in your gradle cache manually.

I likely won't be doing an actual installer, since its just a lot more to maintain when very few will likely ever use it.

Gold872, 2023-12-31:
How would I manually add these to the gradle cache though? There doesn't seem to be a defined location for them

mjansen4857, 2024-01-01:
`C:\Users\Public\wpilib\2024\maven`

Just drop the `com` folder from the build artifact in there.