mjansen4857/pathplanner #532: Add a getter to be able to access the trajectory generated by a PathfindingCommand(holonomic

Ramtech59Sensei, 2023-12-30:
pathfindingCommand has its currentTrajectory as private (I like it because nothing can change it ) .
Creating a getter to access the trajectory generated before the command is executing will be helpful visualization on initial test and  during competition.
Your wiki show the path generated and you visualized it in the image but I cant find a way to access it , One possible way can be to access it in PathFinding.getCurrentPath() but that gives me the latest path calculated and it may not be the correct trajectory .

mjansen4857, 2023-12-30:
https://github.com/mjansen4857/pathplanner/wiki/Java-Example:-Custom-Logging

Ramtech59Sensei, 2023-12-30:
Is that when the Path is running , how can access is before it run ?
Let say I want to generate the path but not run it, still I can get the trajectory ?

Ramtech59Sensei, 2023-12-30:
I am able to see the poses with the link you send when I run autonomous but not before I run it

mjansen4857, 2023-12-30:
There is no path generated until it runs

Ramtech59Sensei, 2023-12-30:
Is the same way for AutoBuilder .followPathWithEvents?
I know we have the path point but can we access the Trajectory to be display on the field2d before run auto?
let say the drivers pick an auto path and I want to show what the robot will do before Auto start

Ramtech59Sensei, 2023-12-30:
I see there is a getTrajectory from Path object , I may be able to use it

mjansen4857, 2023-12-30:
You don't even need that. Just get all the PathPoints from the path with the `getAllPathPoints()` method. Then create a list of poses from each PathPoint's position to put on a Field2d.

Ramtech59Sensei, 2023-12-30:
OK , thanks 
 I was already doing a conversion function to convert from PathPlannerTrajctory.State to WPI Trajectory.State because I realize they trajectories are not compatible any more 
 I will use the function you say

Ramtech59Sensei, 2023-12-30:
you can close this one If you want.
I test the conversion function from state to state, create a trajectory and it work, also should work creating a list of pose2d

Ramtech59Sensei, 2023-12-30:
this is what I did if anyone need it 
`import com.pathplanner.lib.path.PathPlannerTrajectory;
import edu.wpi.first.math.trajectory.Trajectory;
  public Trajectory getStatesToTrajectory(List<PathPlannerTrajectory.State> states){
    List<Trajectory.State> trajStates = new ArrayList<>();
    for(PathPlannerTrajectory.State state : states){
      Trajectory.State trajState = new Trajectory.State(
        state.timeSeconds,
        state.velocityMps,
        state.accelerationMpsSq,
        new Pose2d(state.positionMeters,state.heading),
        state.curvatureRadPerMeter);
          
          trajStates.add(trajState);
    }
    Trajectory traj = new Trajectory(trajStates);
    return traj;
  }`

closed 2023-12-31 by mjansen4857