mjansen4857/pathplanner #533: ono-the-fly path generation freakes out when generating path to "absolute zero" (Pose2d(x: 0, y: 0, rot: 0), GoalEndState(endDegrees: 0 endVel: 0))

TapChap, 2023-12-31:
### heads up
I'm using the term "absolute zero" to refer to a path that the last position given to is an empty Pose2d (0, 0, new Rotation2d(0)), and the GoalEndState endVel = 0, and EndDegrees = 0.

We implemented on-the-fly generation on our robot, from the current robot position to absolute zero, and for the most part, it works, but when the robot is a few centimeters away from the absolute zero (the end of the path) the robot completely freaks out, driving away and spinning uncontrollably...

we ended up terminating the pathFollowingCommand when the odometry says we were below a certain threshold from the desired position, which of course hurts accuracy.

**Expected behavior**
the robot should stop when reaching absolute zero.

**Screenshots**
if that would help, I'll be able to take a video tomorrow

**Versions:**
 - OS: Windows 11
 - GUI Version: [e.g. 2024.0.0]
 - PPLib Version: [e.g. 2024.0.0-beta-6.0]
 - PPLib Language: Java

**Additional context**
link to our code:
https://github.com/ExcaliburFRC/2024RobotTamplate/blob/main/src/main/java/frc/robot/subsystems/swerve/Swerve.java#L336C1-L355C6

Idk if it's relevant but we coded our on-the-fly generation to always set the first position of the path to the current robot location from odometry

so basically event if you call the function like this `swerve.followPath(new Pose2d(6, 7, Rot2d.fromDeg(38)` the actuall path generation wil be like this: `swerve.followPath(odometry.getPose(), new Pose2d(6, 7, Rot2d.fromDeg(38)`

mjansen4857, 2024-01-01:
The code you linked is kinda broken and does not have everything. For instance, you're calling Swerve.followPath in robot container but that method does not exist in your swerve subsystem.

But, the rotation component of on-the-fly path following poses is the direction of travel, not the holonomic rotation. So, that is probably your issue. If you want it to do a straight line to the goal position, you can just set the rotation component to `goalTranslation2d.minus(currentTranslation2d).getAngle()`

TapChap, 2024-01-01:
OMG, the code WAS ok, but in in between me posting this issue and you reading it one of my team member pushed a commit and accidentally deleted that code 😳🥴..

Ill restore it soon..

TapChap, 2024-01-01:
Idiots

TapChap, 2024-01-01:
Should be fine now

mjansen4857, 2024-01-01:
What I said earlier still stands, you can't use the rotation of the robot as the rotation component of poses used for on-the-fly paths. It has to be direction of travel.

TapChap, 2024-01-01:
I'm not sure I got it, does this look right?
https://github.com/ExcaliburFRC/2024RobotTamplate/blob/main/src/main/java/frc/robot/subsystems/swerve/Swerve.java#L347C3-L361C11

TapChap, 2024-01-02:
Ight, cool, it works. Thanks

TapChap, 2024-01-02:
I've done what you said, and now whenever I enter any setpoint OTHER then absZero, it works perfectly and drives there..
Besides absZero, in which it it first drives there, and when it basically gets there it freaks out completley..

I'll shoot a video of the robot driving to (x=0.1, y=0, deg=0) and (x=0, y=0, deg=0) and upload it here soon.

TapChap, 2024-01-02:
This is the current code I'm running

```
    // on-the-fly auto generation functions
    public Command followPath(double endVel, double endDegrees, Pose2d... positions){
        return AutoBuilder.followPathWithEvents(
                        new PathPlannerPath(
                                PathPlannerPath.bezierFromPoses(positions),
                                PATH_CONSTRAINTS,
                                new GoalEndState(endVel, Rotation2d.fromDegrees(endDegrees))));
    }

    // drives the robot in a straight line from current location to a given Pose2d
    public Command robotToPose(Pose2d position, double endVel){
        return followPath(endVel, position.getRotation().getDegrees(), getStraightLinePoses(position.getTranslation()));
    }

    private Pose2d[] getStraightLinePoses(Translation2d setpoint){
        Translation2d current = odometry.getEstimatedPosition().getTranslation();
        Rotation2d directionOfTravel = setpoint.minus(current).getAngle();

        return new Pose2d[]{
                new Pose2d(current, directionOfTravel),
                new Pose2d(setpoint, directionOfTravel)
        };
    }
    // ----------
```

TapChap, 2024-01-02:
Here are the videos, I should Point out that the x=0.1 / x=0 is the ONLY change between the two videos, (in the code and other aspects)

the 0,0 point of the odometry could be seen on the ground with the 4 tape marks...

links to the videos:
https://drive.google.com/file/d/1DTMBpjJnYuLnwYrBqfOwA82cnyPyR0Ai/view?usp=sharing
https://drive.google.com/file/d/1DM5jZlXem45M_P9XNr30R-RKliAIbxBN/view?usp=sharing

mjansen4857, 2024-01-02:
You're creating the command when robot code starts, not when you press the button. So, the position used as the "current" will be (0, 0). Meaning that you are generating a path from (0, 0) to (0, 0). This causes all of the points of the curve to be the same, which breaks things. The reason the robot still gets there is because replanning is on, but then it rejoins the busted path and freaks out. You need your button to run a command that will create the path, then schedule the path following command:

```java
// This creates a path when code starts
//controller.cross().onTrue(swerve.robotToPose(new Pose2d(0.1, 0, Rotation2d.fromDegrees(0)), 0));

// This creates a path when the button is pressed
controller.cross().onTrue(Commands.runOnce(() -> swerve.robotToPose(new Pose2d(0.1, 0, Rotation2d.fromDegrees(0)), 0).schedule()));
```

TapChap, 2024-01-02:
That makes sense... Ill check it soon, BTW i better way to do it is to use a [proxyCommand](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/wpilibj2/command/ProxyCommand.html)

mjansen4857, 2024-01-02:
As long as you use the constructor that takes in a supplier, not the command directly

Starlight220, 2024-01-02:
DeferredCommand (new for 2024) could be better for this purpose, as it doesn't introduce the proxy semantics.

TapChap, 2024-01-03:
ok, this seems to work, Thanks!

closed 2024-01-03 by TapChap